EmuELEC/EmuELEC #495: FAILURE: scripts/install PPSSPPSDL has failed!

dovart, 2021-02-26:
Can't build EmuELEC dev branch.

The issue is with the PPSSPPSDL and libpng.

-- Could NOT find PNG: Found unsuitable version "1.6.36", but required is at least "1.7" (found /EmuELEC/build.EmuELEC-Amlogic.aarch64-4.0/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/lib/libpng.so)
CMake Error: File /EmuELEC/build.EmuELEC-Amlogic.aarch64-4.0/PPSSPPSDL-39198b53a521ae7d8031303173b1b12164659017/ext/miniupnp/miniupnpc/miniupnpcstrings.h.cmake does not exist.
CMake Error at CMakeLists.txt:1894 (configure_file):
  configure_file Problem configuring file

shantigilbert, 2021-02-26:
Read the error "Could NOT find PNG: Found unsuitable version "1.6.36", but required is at least "1.7"

shantigilbert, 2021-02-26:
update this package: https://github.com/EmuELEC/EmuELEC/blob/dev/packages/graphics/libpng/package.mk

closed 2021-02-26 by shantigilbert

ws02589111, 2021-03-13:
-- Found PNG: /home/liwx/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.0/toolchain/aarch64-libreelec-linux-gnueabi/sysroot/usr/lib/libpng.so (found suitable version "1.7.0beta89", minimum required is "1.7") 
CMake Error: File /home/liwx/EmuELEC/build.EmuELEC-Amlogic.aarch64-4.0/PPSSPPSDL-39198b53a521ae7d8031303173b1b12164659017/ext/miniupnp/miniupnpc/miniupnpcstrings.h.cmake does not exist.

The libpng 1.7 version is only found beta，the version check can not pass

ws02589111, 2021-03-14:
The issue  maybe relate to https://github.com/hrydgard/ppsspp/pull/12748

miwasp, 2021-03-15:
I just made a complete fresh clone, checkout and make of dev and get exactly this error.
The package.mk of libpng loads the latest stable version 1.6.37 but PPSSPPSDL somehow wants version 1.7.

Additionally there are two versions of libpng in the build directory, the above mentioned 1.6.37 and another one with version 1.6.36?

miwasp, 2021-03-16:
I did it all again from scratch and now it compiled!?
I don't know what went wrong last time.

ws02589111, 2021-03-17:
@shantigilbert 
Any suggestion？

ws02589111, 2021-03-19:
> I did it all again from scratch and now it compiled!?
> I don't know what went wrong last time.

what branch you compiled?  dev？

miwasp, 2021-03-19:
> what branch you compiled? dev？
Yes, i compiled dev. In the meantime i did it again from scratch and it worked again.

ws02589111, 2021-03-21:
> > what branch you compiled? dev？
> 
> Yes, i compiled dev. In the meantime i did it again from scratch and it worked again.

Finally， I reinstalled my system to solved the problem，I don't know why after `make clean`  to run `make image` is no use