EmuELEC/EmuELEC #497: V4.0 Samba server not work normal , V3.9 no problem on it

RitoKatsu, 2021-02-27:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - Version [3.9  4.0]

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: [S095x3]
 - Brand: [h96]
 - Model [max x3}
 - Version [4g ram /128 G]

**Additional context**
Add any other context about the problem here.

Hello
I have a H96 max X3 TV-BOX and use Emuelec V3.9 ,
It's normally to copy files from windows 10 PC to TV-BOX with SAMBA.

Now I make new TF Card with V4,0 [ user sm1_s905x3_4g_1gbit.dbt ]
There are 2 partition [ FAT32 ] , that's good idea from that trasfer file with PC always.

But the SAMBA funtion seems not work properly
I can see folders on file manager , but when I paste files ,
notify message show ( disk space no enough ..... ) 

抱歉 英文不好 怕說明的不是很正確
以下我用中文在表達一次

先前我在H96 max X3電視盒使用過Emuelec V3.9這個版本
然後透過Windows 10網路芳鄰的方式將檔案拷貝到emuelec系統中
過程順利沒發生什麼問題

這幾天系統提示有4.0版本可以使用
所以我用了另外一塊TF卡做了V4.0版的Emuelec系統
插入H96 max X3後操作上沒什麼不同
但是當我要透過網路芳鄰傳送檔案的時候
會出現"檔案'  xxxxxx-xxxx   '太大無法放置在目的地檔案系統
(PS:我放的檔案其中有一個 .srm 才 16K 也放不進去)

這次的改版在改善複製檔案的便利上很棒
但是否能讓 SAMBA 功能 跟 FAT32 partition 功能 同時並存?

謝謝

shantigilbert, 2021-02-28:
I cannot reproduce this.

shantigilbert, 2021-03-05:
Closing this, feel free to reopen with more information, as I cannot reproduce it.

closed 2021-03-05 by shantigilbert