EmuELEC/EmuELEC #498: Please add support in your builds for LibreComputer LaFrite 1GB an 512MB if possible.

devegoo, 2021-02-27:
Hi. I have this SBC in 2 versions 512mb and 1gb ram and i have buggy test build for 1gb ver.
forum topic:
[https://forum.loverpi.com/discussion/1185/emuelec-image-for-lafrite#latest](url)
downlload location:
[https://forum.loverpi.com/home/leaving?target=https%3A%2F%2Fdrive.google.com%2Fdrive%2Ffolders%2F12ZvLhi7S86YubAho04JXsiqG129r25xW%3Fusp%3Dsharing](url)
this build can be upgraded to 4.0 by menu easy
by now dev  branch for Amlogic-ng supports only Odroid N2,C4,HC4,Generic
LaFrite Is included in this branch but in "EmuELEC/projects/Amlogic-ng/options" default by now at the end of file not have LaFrite LePotato something like this must be included:
 # CoreELEC Subdevices
    SUBDEVICES="......................LePotato LaFrite"
BUGS:
1.games starts fine in retroarch mode but not in emulsation mode which is bug , and i can't resolve it manually....

Please add support for this devices....
Best Wishes.
T.S.

shantigilbert, 2021-02-27:
It is not a bug, Lafrite does not have enough RAM to be playable with ES, specially not the 512mb, 1gb of ram will work, but 2gb of ram minimum is recommended.

devegoo, 2021-02-27:
ok i have now only one option for booth versions for gaming&: Libreelec nightly and playing via addons or maybe some other systems designed fr gaming purposes have not such bigger RAM requirements, some developer ported batocera linux to  Amlogic box S812 but this MXQ also have  2gb onboard
[https://forum.libreelec.tv/thread/21798-meson-8-lives/?postID=149974#post149974](url)

closed 2021-02-27 by devegoo

shantigilbert, 2021-02-27:
EmuELEC will work if you boot into Retroarch, but it will be limited