EmuELEC/EmuELEC #499: EmuElec 4.0 only boot from external SD card reader

eric4gith, 2021-02-28:
Hi, 
Hope I am posting in the right place.

I have one X96 Max+ (S905X3/4G/32G/Gbe) and one T95Z plus (S912/3G/32G/Gbe). MY SD card is a Samsung 8GB Class 2 SDHC.

On T95Z Plus, EmuElec 3.9 (using gxm_q201_3g_1gbit.dtb) can boot T95Z Plus from internal card reader without problem. 

On X96 Max+, EmuElec 4.0 (using sm1_s905x3_4g_1gbit.dtb) can boot from "external" card reader (plug onto USB). However, same SD card, no success boot from internal card reader. I got error message "Could not mount LABEL=EMUELEC ***".

I have tried EmuElec 3.9 from internal card reader of X96 Max+, still got the same error message. On external card reader, the boot is successful.

I guess, either the internal card reader of X96 Max+ is crapy or there is a compatible issue. Just providing some facts I found and  hoping it can be fixed in the future.

eric4gith, 2021-02-28:
I Just tried CoreELEC 9.2.6 on X96 Max+ (using "sm1_s905x3_4g_1gbit.img"). Still got error message "Could not mount LABEL=COREELEC ***". So, I guess the problem is inherited from CoreELEC, maybe I should report my problem or find a solution in CoreELEC forum.

shantigilbert, 2021-02-28:
The issue is neither CoreELEC nor EmuELEC is on your side, use another SD card, another software to flash it (etcher, or win32imager or another) or try a different SD card reader.

closed 2021-02-28 by shantigilbert