EmuELEC/EmuELEC #500: Build errors when building SDL - Incorrect checksum on one URL, 404 on the other URL

Dwedit, 2021-02-28:
I am trying to build the distro for the GameForce handheld, using the command `PROJECT=Rockchip DEVICE=GameForce ARCH=aarch64 make image`

At some point when trying to build the distro, it wants to build SDL, so it downloads the source code.

First it tries https://hg.libsdl.org/SDL/archive/74055dd839e4.tar.gz, which is a working link, but it complains that the checksum is incorrect:
>Incorrect checksum calculated on downloaded file:
>got c689d626ab6fbc0ed0bd03a49aad6b0af483a6cd5c352cd130e177587e0f97c1
>wanted 809644ef6e2050148b6649eca395e8472e66b9f5ef6d20392baf5e7fb178160a

Then it tries http://sources.libreelec.tv/mirror/SDL/SDL-74055dd839e4.tar.gz, which is 404.

Then it repeatedly downloads from the first URL a total of 10 times, each time rejecting the file because it has an incorrect checksum.

I will try editing the expected checksum in `/EmuELEC/packages/sx05re/tools/SDL/package.mk` on my PC.

shantigilbert, 2021-02-28:
don't take this the wrong way... but 2 things:

1.- How are you expecting to compile something if you are not willing to at least put a bit of effort on investigating issues?
2.- Do a quick search on GH issues, there has already been several with the same problem, and not to mention the issue is also fixed on the dev branch 14 days ago.

https://github.com/EmuELEC/EmuELEC/issues/459
https://github.com/EmuELEC/EmuELEC/issues/489
https://github.com/EmuELEC/EmuELEC/commit/0ec9e9acd2e672e7bf63025beec1906867247239

closed 2021-02-28 by shantigilbert

Dwedit, 2021-02-28:
I had looked through the open issues page for the word "SDL", and saw nothing before I posted this.  I wouldn't expect to find active issues that were in a closed state.

shantigilbert, 2021-02-28:
Thats because its not an active issue... like I said, it was fixed 14 days ago

Dwedit, 2021-02-28:
So it was fixed in the "dev" branch, but not the "master" branch, that's a little confusing, but I understand.

shantigilbert, 2021-02-28:
Nothing is commited to the "master" branch as it's "frozen" to the current "stable" version (4.0 in this case) no fixes are pushed until a new "stable" version is released, everything is worked on the "dev" branch, this includes fixes and changes. 

Compiling the master branch is the same as just using the already made images available on "releases"

shantigilbert, 2021-02-28:
I modified the WiKi to reflect this: https://github.com/EmuELEC/EmuELEC/wiki/Developers-corner---How-to-submit-a-pull-request-(PR)

shantigilbert, 2021-02-28:
I just noted you are trying to build for the GameForce, so there would be no release file available for it yet, I apologize. but still "dev" is the branch you should be compiling

Langerz82, 2021-09-05:
Is the retrieved checksum ok? Because I just overwrited it in the package.mk. I'm compiling so guess I will know soon ha.