EmuELEC/EmuELEC #501: DS3 gamepad can't connect BT in EmuELEC 4.0

eric4gith, 2021-02-28:
I tried using DS3 controller in EmuELEC 3.9 on T95Z Plus. Both wired and wireless connection is working.

Then I tried it in EmuELEC 4.0 on X96 Max+. Wired connection is working, but the system does not seem to update the BT address. There should be an BT address item in "FORGET A BLUETOOTH CONTROLLER" setting. That's what I did see in EmuELEC 3.9 on T95Z Plus.

Not sure whether it's a bug of EmuELEC 4.0 or X96 Max+ hardware.

eric4gith, 2021-02-28:
I just tried EmuELEC 3.9 on X96 Max+. The result is the same with EmuELEC 4.0. No BT address update.

shantigilbert, 2021-02-28:
Please do a search, there has been several issues with the X96, its a POS the BT chip is not supported, there are no dirvers available. 

I am not sure why you said it was working on 3.9 it is not.

closed 2021-02-28 by shantigilbert

eric4gith, 2021-02-28:
Sad to hear that.
Let me make it more clear. I said BT working on EmuELEC 3.9 on T95Z Plus, not X96 Max+. In fact, BT on X96 Max+ both EmuELEC 3.9 & 4.0 is not working. It match the BT chip problem you mentioned.

Just got  a log file. Maybe too late. http://ix.io/2RdD

shantigilbert, 2021-02-28:
OK yeah I apologize, I just read t95 or x96 and I assume they all the same, as they seem to be the ones that give a lot of issues. 

anyways, the issue is the same, the BT chip is not supported as it has no drivers or firmware that I can use.