EmuELEC/EmuELEC #502: [BUILD] Package tic-80 gives build errors at CMake step

Dwedit, 2021-02-28:
**Describe the bug**

When building the distro, I am getting an error that tic-80 has failed to build.

The error is happening at the CMake step.  It is running CMake on a file called `~/EmuELEC/build.EmuELEC-GameForce.aarch64-4.0/tic-80-f4688a5af41c6e4f052443cdd42fe0061693f1d6/CMakeLists.txt` and it gives an error at line 505, complaining that it doesn't contain a `CMakeLists.txt` file.

CMakeLists.txt contains this:
```
################################
# ZIP
################################

set(CMAKE_DISABLE_TESTING ON CACHE BOOL "" FORCE)
add_subdirectory(${THIRDPARTY_DIR}/zip)     <- Line 505
```

`$(THIRDPARTY_DIR/zip)` evaluates to ` /home/dan/EmuELEC/build.EmuELEC-GameForce.aarch64-4.0/tic-80-f4688a5af41c6e4f052443cdd42fe0061693f1d6/vendor/zip`

When I look inside that directory, I see just one file, named `.git`. 
There is no `CMakeLists.txt` file that it is looking for.
The lone `.git` file contains this text: `gitdir: ../../.git/modules/vendor/zip`

----------

I went to the directory `/home/dan/EmuELEC/build.EmuELEC-GameForce.aarch64-4.0/tic-80-f4688a5af41c6e4f052443cdd42fe0061693f1d6/vendor/zip`, and typed in "git status"  I am seeing this:
```
On branch master
Your branch is up-to-date with 'origin/master'.
Changes to be committed:
  (use "git reset HEAD <file>..." to unstage)

        deleted:    .github/workflows/build.yml
        deleted:    .gitignore
        deleted:    CMakeLists.txt
        deleted:    CONTRIBUTING.md
        deleted:    Doxyfile.in
        deleted:    README.md
        deleted:    UNLICENSE
        deleted:    cmake/Config.cmake.in
        deleted:    cmake/FindASan.cmake
        deleted:    cmake/FindMSan.cmake
        deleted:    cmake/FindSanitizers.cmake
        deleted:    cmake/FindTSan.cmake
        deleted:    cmake/FindUBSan.cmake
        deleted:    cmake/asan-wrapper
        deleted:    cmake/cmake_uninstall.cmake.in
        deleted:    cmake/sanitize-helpers.cmake
        deleted:    src/miniz.h
        deleted:    src/zip.c
        deleted:    src/zip.h
        deleted:    test/CMakeLists.txt
        deleted:    test/test.c
        deleted:    zip.png
```

It says that every file is "deleted".

shantigilbert, 2021-02-28:
https://github.com/EmuELEC/EmuELEC/commit/d03efaa1cdf6c91086758c1a64c36fb3656b9a47

shantigilbert, 2021-02-28:
Like I told you before, don't build `master`, build `dev`

closed 2021-02-28 by shantigilbert

Dwedit, 2021-02-28:
Hmm, I thought I was building dev, I did the git command to switch the repository.