EmuELEC/EmuELEC #503: Not launching Sonic/Genesis games

granspuny, 2021-03-01:
Hi all,

I am finding trouble to execute Genesis/MegaDrive games in general. I read somewhere that for these games the BIOS is not needed, so I am just simply copying the ROMs in the correct path but when executing the OS they are not shown. **Even the built-in Sonic games do not work: when I try to run them, the screen fades to black, and goes back to the menu**.

Games for the rest of the platforms are working fine.

I looked at the known issues section, and made sure that no two game pads are connected, but that was not the case.

This is the release I am using: EmuELEC-Amlogic-ng.aarch64-4.0-Generic.img.gz

And my TV Box is a x96 Max+ (2GBs) with Amlogic S905X3 SoC.

Thank you very much and regards,

Spuny

dm2912, 2021-03-01:
> >Even the built-in Sonic games do not work: when I try to run them, the screen fades to black, and goes back to the menu.

Have you installed the data files for the Sonic ports? They wont run without them.

instructions here https://cloud.dimspace.xyz/s/Tn37bYqwdF5k9sa

granspuny, 2021-03-01:
> > > Even the built-in Sonic games do not work: when I try to run them, the screen fades to black, and goes back to the menu.
> 
> Have you installed the data files for the Sonic ports? They wont run without them.
> 
> instructions here https://cloud.dimspace.xyz/s/Tn37bYqwdF5k9sa

Thank you very much for your advice. I follow your instruction and it worked!

Also, I tried again copying the Genesis ROMs and this time they were detected correctly. I guess I did something worng the first time but don't know what it was tbh.

My suggestion is to include the Sonic games in the section Needed BIOS (https://github.com/EmuELEC/EmuELEC/wiki/bios), just like Doom, Cave Story and the rest of the ports. Otherwise you don't know that you need an external file to make them work.

Thanks again and regards,

Spuny

closed 2021-03-01 by shantigilbert