EmuELEC/EmuELEC #504: Emuelec for X96 Air S905X3 not working

adamfaizz, 2021-03-02:
Hi,

I have tried installing version 3.4 upto 3.9. So far only managed to install 3.4 and 3.5 on  X96 Air S905X3.

However, in both installations, I got a blank screen after restart or start again after the device is shutdown.

What could be the issue? I really want EMUELEC for my box.

Really appreciate the assistance.

AlienatedDiego, 2021-03-02:
Have you put the correct .dtb file in the root of the sd and renamed it to dtb.img?

SamuEDL, 2021-03-02:
Why not install EmuELEC 4.0? I have an x96 AIR (P3) with EmuELEC 4.0 and no problem. Also in the past I used EmuELEC 3.x and I had no bugs either.

Are you using "EmuELEC-Amlogic-ng.aarch64 - *. * - Generic.img.gz" and the dtb corresponding to x96 air?

adamfaizz, 2021-03-05:
> Why not install EmuELEC 4.0? I have an x96 AIR (P3) with EmuELEC 4.0 and no problem. Also in the past I used EmuELEC 3.x and I had no bugs either.
> 
> Are you using "EmuELEC-Amlogic-ng.aarch64 - *. * - Generic.img.gz" and the dtb corresponding to x96 air?

Here are the steps that I did for v4.0

1. Burn image EmuELEC-Amlogic-ng.aarch64-4.0-Generic into SD card
2. Copy sm1_s905x3_4g.dtb to root and rename to dtb.img (Also tried sm1_s905x3_4g_1gbit.tdb but same result)
3. Boot into recovery mode -Emuelec installation - Resizing partition - done Resizing file system - done, Formatting  EEROMS partition - done, Rebooting in 5s...now and the screen just hang.

For v3.4 and 3.5, I managed to complete the installation, got up game pad setup. then shutdown. When I tried to boot again, or restart after the gamepad setup, it just displays blank screen.

adamfaizz, 2021-03-05:
> Have you put the correct .dtb file in the root of the sd and renamed it to dtb.img?

yeah I did

SamuEDL, 2021-03-05:
> Here are the steps that I did for v4.0
>
> 1. Burn image EmuELEC-Amlogic-ng.aarch64-4.0-Generic into SD card
> 2. Copy sm1_s905x3_4g.dtb to root and rename to dtb.img (Also tried sm1_s905x3_4g_1gbit.tdb [...]

Is the SD of a recognized brand? (Sandisk, samsung and etc) Since those errors, they seem more of a poor quality SD.

adamfaizz, 2021-03-09:
> Is the SD of a recognized brand? (Sandisk, samsung and etc) Since those errors, they seem more of a poor quality SD.

I am using Kingston 64GB class 10.. is it ok?

adamfaizz, 2021-03-09:
> I am using Kingston 64GB class 10.. is it ok?

also tried sandisk.. both are new sd cards.. same result

suspect there is something with the box's firmware... can you give me your firmware version?

SamuEDL, 2021-03-09:
> suspect there is something with the box's firmware... can you give me your firmware version?

If you ask the Android firmware that I have installed. Currently I do not use the stock rom, but slimbox.
P.D: Although when I had the stock rom, I also had no problem with EmuELEC.

closed 2021-04-01 by shantigilbert