EmuELEC/EmuELEC #505: No sound on X96 Max+ S905x3 with Emuelec 4.0

durstboiler, 2021-03-04:
**Describe the bug**
No sound on emulationstation, retroarch or PPSSPP, HDMI or AV.

**Expected behavior**
I don't have any sound anywhere. The games work but without sound.

**EmuELEC version:**
v4.0

**Device
 - SOC: S905x3
 - Brand: OTT TV BOX
 - Model X96 Max+
 - Version 2g 16g

shantigilbert, 2021-03-05:
Steps to reproduce?

durstboiler, 2021-03-05:
In the emuelec log I saw a line with an audio driver initialization problem. But this line is no longer displayed ...

shantigilbert, 2021-03-05:
Well... if you changed the audio driver, what do you expect to happen? please reproduce this on a clean image, as I cannot reproduce this, all the games have no problems with sound.

durstboiler, 2021-03-05:
But you can imagine that I changed the audio driver because it was not already working ...

shantigilbert, 2021-03-05:
No, I could not, since you did not mentioned that. and Like I said I cannot reproduce this, so try with a clean image

durstboiler, 2021-03-05:
I reinstalled emuelec v4, very clean image, same result, no sound. I have tried the other dtb S905x3 but only the one mentioned above works.

shantigilbert, 2021-03-05:
are you using HDMI?

durstboiler, 2021-03-05:
Yes ! Big thanks. I hadn't seen this option, sorry.

shantigilbert, 2021-03-05:
> Yes ! Big thanks. I hadn't seen this option, sorry.

is it working now? what option worked for you?

durstboiler, 2021-03-05:
Yes it working, it's Audio 0.1.

closed 2021-03-05 by shantigilbert