EmuELEC/EmuELEC #507: more portable shell scripts

zhangn1985, 2021-03-05:
https://github.com/EmuELEC/EmuELEC/blob/1270299916118c5b6e246ec41665dce6b31dc74e/packages/sx05re/emuelec/config/emuelec/scripts/emuelecRunEmu.sh#L70-L81

https://github.com/EmuELEC/EmuELEC/blob/1270299916118c5b6e246ec41665dce6b31dc74e/packages/sx05re/emulationstation-addon/config/es_systems.cfg#L118

`/emuelec/scripts/emuelecRunEmu.sh FBN %ROM% -PCPS2`

why do you design a parameter this way: `-PCPS2`, usually it can be `-P CPS2` and easy use getopts/getopt, or even if python's argpaser module.

the way uses the script, make the argparser very difficult.

in this case, CORE="FBN %ROM% -PCPS2" this is a error....

would you like change the scripts and make it more portable and readable?

shantigilbert, 2021-03-05:
Its made this way because that is the way it worked for me, I don't see why it needs to be portable, as this script is not being used anywhere else but on EmuELEC? 

But if you have a better solution Pull requests are always welcomed.

zhangn1985, 2021-03-05:
I don't want to hide something, these scripts only used in this project. but I'm trying to port/rebase EmuELEC to LibreELEC, thus to support mainline opensource software stack.

the first step, I make Lakka work on LibreELEC, I test it, and works well, 2048 and FBNeo games works.
based on that, I start EmuELEC, the UI launches, but no game works, including RA games. thus I try to understand how es launches games, then find the scripts. es uses these scripts to lauch game, and the command line is in these config file, right?

I try launch 2048, use the scrpit, but get lots of errors. and look to the command line in conf file, I really don't want to make so big downstream changes. thus ask if you also want to make changes to the scripts.

shantigilbert, 2021-03-05:
I compiled a LE (kernel 5.x) version privately and found no issues with the scripts.

If you know of a better way, please make the changes and open a pull request.

zhangn1985, 2021-03-08:
I know why I get error when porting to LE, 2048 launch script in default gamelist.xml for EmuELEC is can't launch 2048.
maybe I uses `busybox bash` as `bash`...

just close it.

closed 2021-03-08 by zhangn1985

shantigilbert, 2021-03-08:
bash (real bash, not busybox) is needed for the script