EmuELEC/EmuELEC #514: Poor Flycast performance on box s905x3

durstboiler, 2021-03-08:
I noticed a slight decrease in performance with Flycast on S905x3, emuelec v3.9 or v4.0, compared to my other box s905x 1g 8g, however in 32bits and has much lower characteristics.
For the moment, the only other tested emulator is PPSSPP, and on this one the performances are much better this time.

SOC: S905x3
Brand: OTT TV BOX
Model X96 Max+
Version 2g 16g

guy2010, 2021-04-04:
Which dtb file did you use? I have an X96 +, 4gb model   thanks

Cee123, 2021-05-12:
Do you have threaded rendering turned off? If so, then turn that on. Also turn off Synchronous Rendering. I don't know if this would help, but hope it does.

AlienatedDiego, 2021-05-12:
This is highly dependent on the game, some get better and some don't. In 4.1 core flycast32 (32bit) was added for this. The only game that I have noticed got worse with the jump to 64bit is R.E Code Veronica, but I consider it an exception. In general terms the performance is the same and in some cases it improves.

closed 2021-06-07 by shantigilbert