EmuELEC/EmuELEC #515: Bluetooth and Wifi Drivers to X96 MAX PLUS new version

elwaskop, 2021-03-08:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Hello good Morning
Will it be possible to add the drivers for the new hardware that is coming is the x96 max plus? I have tried to find the model of the wifi and bluetooth chip and through applications it appears AR6K_WLAN, from what I have read it refers to a chip for Atheros. I do not know if you know it and can include it, thank you very much and I thank you for all your effort.

AlienatedDiego, 2021-03-08:
Hi, first you should provide the chip model (probably need to open the box for this) and check if there are drivers for it.

Iamghostfor, 2021-03-09:
> Hi, first you should provide the chip model (probably need to open the box for this) and check if there are drivers for it.

Hello ,I found the wifi and bluetooth chip is rtl8822cs（both of them is one chip）；

and is this it driver? https://github.com/chewitt/RTL8822CS

elwaskop, 2021-03-09:
> > Hi, first you should provide the chip model (probably need to open the box for this) and check if there are drivers for it.
> 
> Hello ,I found the wifi and bluetooth chip is rtl8822cs（both of them is one chip）；
> 
> and is this it driver? https://github.com/chewitt/RTL8822CS

Hello friend, first of all I thank you very much for your prompt response. Really I have not dared to open the tvbox for warranty reasons. Is there a way to test the driver? Or find out the chip through software? I have used HWINFO.apk and I only got AR6L_WLAN and RTL8211F ethernet. In case you tell me that there is no option then I will open it to verify.
 :)
Greetings

closed 2021-04-01 by shantigilbert