EmuELEC/EmuELEC #516: Cannot pair a new device via Bluetooth in v4.1-TEST-02262021

mrabssalom, 2021-03-08:
**Describe the bug**
After updating v4.0 (bug fixed) to latest Test Version, my device doesnt search for new bluetooth controlers.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Emuelec menu'
2. Click on 'Pair a new bluetooth device'
3. A message "Searching for devices" appears but last only a few seconds and dont found anything.

**Expected behavior**
Previously message "Searching for devices" lasted arround a minute or two and devices were found.

**EmuELEC version:**
 v4.1-TEST-02262021

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: s922x
 - Brand: Bellink
 - Model GT King
 - Version RevB

shantigilbert, 2021-03-17:
Thanks for the report I will try to reproduce on my Gt-King

shantigilbert, 2021-03-18:
so on v4.0 it works and on the same box on 4.1 does not?

shantigilbert, 2021-03-18:
Can you try the newest test?

mrabssalom, 2021-03-18:
> Can you try the newest test?

Will try this afternoon. 

v4.0 works perfectly, with that test version i try to pair a new bluetooth device and it doesnt search for nearby devices (message last 2 seconds and then fades out)

shantigilbert, 2021-04-01:
After 14 days no information, will close this, feel free to reopen if the issue persists

closed 2021-04-01 by shantigilbert