EmuELEC/EmuELEC #520: Get some building error in lastest dev branch

ws02589111, 2021-03-11:
System: ubuntu server 18.04 (get image from http://releases.ubuntu.com/18.04/) in vmware

Version: dev branch, commit id 104846af418430adc4ca29f147cd4e884a53b86b

Command:
`sudo apt update && sudo apt upgrade
sudo apt-get install gcc make git unzip wget xz-utils libsdl2-dev libsdl2-mixer-dev libfreeimage-dev libfreetype6-dev libcurl4-openssl-dev rapidjson-dev libasound2-dev libgl1-mesa-dev build-essential libboost-all-dev cmake fonts-droid-fallback libvlc-dev libvlccore-dev vlc-bin texinfo premake4 golang libssl-dev curl patchelf xmlstarlet default-jre

sudo apt-get install tzdata xfonts-utils lzop

git clone https://github.com/shantigilbert/EmuELEC.git EmuELEC    
cd EmuELEC  
git checkout dev
PROJECT=Amlogic ARCH=aarch64 DISTRO=EmuELEC make image `

1.
[image]

After run `sudo apt-get install xsltproc` the error  is resolved, maybe add xsltproc install to readme?

2.
[image]

configure: error: ncurses selected, but library not found
FAILURE: scripts/install linux has failed!
[064/452] [FAIL] install linux

I tried to run `sudo apt-get install libncurses5-dev libncursesw5-dev` but no effect

Maybe the issue is relate the commit #7733f6674caddab2878579ebb82731576e02f8cc  ?

shantigilbert, 2021-03-11:
You need to find out what package is giving you the error, and then add `ncurses` as a dependency.

shantigilbert, 2021-03-12:
fixed on this commit: https://github.com/EmuELEC/EmuELEC/commit/368aa8e0d45213ad6cd26c8fafd98cc8b6dc2ea3

ws02589111, 2021-03-12:
OK，thanks，the issue solved

closed 2021-03-12 by ws02589111