EmuELEC/EmuELEC #523: dtb request for tx3 mini with wifi working .(THIS IS NOT A COMPLAINT , BUT A REQUEST HELP)

bluedoom47, 2021-03-15:
hello sir,
you have made a very good softwere ,and we are very gratefull for it. thank you .
my problem is that i have a tx3 mini a , which in emuelec runs everything but wifi. iv  used the dtb given in device tree but wifi dosent work. if i can give you the the firmware link that is working android firmware with working wifi , can you extract the dtb from there please?

dm2912, 2021-03-15:
which dtb are you using? because my tx3-mini's wifi worked without any problems and its probably one of the most widely used boxes out there.

dm2912, 2021-03-15:
may be that the new release of the mini (possible a fake clone) has different wifi https://discourse.coreelec.org/t/tx3-mini-no-wifi-and-no-vfd-display/14912/6

SamuEDL, 2021-03-17:
If we talk about the tanix tx3 mini (s905w) with wifi 9082xs. It will never work there is no source code or driver. Not the fault of the dtb.
https://discourse.coreelec.org/t/wifi-problem-on-tanix-tx3-mini/9547

dm2912, 2021-03-17:
> If we talk about the tanix tx3 mini (s905w) with wifi 9082xs. It will never work there is no source code or driver. Not the fault of the dtb.
> https://discourse.coreelec.org/t/wifi-problem-on-tanix-tx3-mini/9547

yeh, that thread is not accurate. My first emuelec box was the TX3 Mini and the WIFI 100% worked and Tanix made everything for the original TX3 Mini known on Freaktab forums.

But.. there appears to be either

a) A new revision of the Mini using a differing wifi chip
or
b) a bunch of clones that are going out that have different parts. (more likely)

closed 2022-01-21 by shantigilbert

dyorgio, 2022-03-06:
Hello, I guess that I found a source (for 9012 aka 9086):

https://github.com/hichard/my_wifi_driver/tree/master/trunk_driver

dyorgio, 2022-03-21:
Guys, we finally get it works and stable: https://github.com/CoreELEC/smartchip 
Now, we just need to wait EmuELEC based on CoreELEC 19.5 (or 19.6 😂).

shantigilbert, 2022-03-21:
> CoreELEC 19.5 (or 19.6 joy).

EmuELEC is based on CE 20