EmuELEC/EmuELEC #524: s905x3 Android TV box - No internet connection (wifi or ethernet) / no bluetooth

8bit16bit, 2021-03-17:
Hi, I've recently got a s905x3 Android TV box for use with EmuELEC, easy to use and install on the MicroSD card, everything works great except for a few things:

1. No internet at all - On the TV box itself (without the EmuELEC MicroSD inserted) the internet connection has no problem whatsoever, so I know the box itself is working perfectly. But when booted in to EmuELEC I cannot connect to the internet at all, either by wifi or a wired ethernet connection. I've tried rebooting, restarting, wifi on, wifi off, booting in to RetroArch - nothing works.

2. Cannot connect gamepad with bluetooth - I suspect this is perhaps part of the same problem as #1? Maybe a driver issue, I'm not sure.

3. I've had a few games crash the system to a red screen with white text - Not so unusual for emulation, but at this screen my (wired) gamepad stops responding, so I have no other option but to pull the plug out to turn off the machine. I see a 'Close' button at the bottom of the screen, but have no way to interact with it. So my question is, is pulling the plug out of the socket my only course of action here, or is there a safer way to restart/shutdown once I get hit with this 'red screen of death' ?

Thanks for your help.

AlienatedDiego, 2021-03-17:
What box is it, what are the specs of your device, what version of EmuELEC?  
you may not be using the correct .btb or the bt / wifi chip is not supported  
the log screen issue only occurs with some gamepads, it may be fixed in 4.1; you can try another gamepas or connect a keyboard to escape the death screen

8bit16bit, 2021-03-17:
- Box is HK1 Box - https://www.amazon.co.uk/Android-Amlogic-S905X3-Netflix-Youtube/dp/B0852YS4ZM/
- EmuELEC is latest version, 4.0
- .dtb I used was the version for S905x3 boxes, I was following the install guide on the wiki - although is there a way to check as I'm only 95% sure, I could've also used the generic S905 version instead, but almost certain I used the one with .ng in the filename.

I'll try plugging in a keyboard next time I get the red death screen, thanks for that.

AlienatedDiego, 2021-03-17:
according to the specs that shows the correct .dtb it would be sm1\_s905x3\_4g\_1gbit.dtb

8bit16bit, 2021-03-18:
That worked thank you. I realised looking at the file sizes that the dtb I used was "sm1_s905x3_4g.dtb" - when I changed it to the 1gbit one it worked immediately. I feel silly for making such a rookie mistake, thanks again! :)

AlienatedDiego, 2021-03-18:
Glad to hear it worked .happy retrogaming 🙂

closed 2021-04-01 by shantigilbert