EmuELEC/EmuELEC #531: Problems with using the B button in games

guy2010, 2021-03-21:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

guy2010, 2021-03-21:
[image]
The game crashes.

AlienatedDiego, 2021-03-21:
Do you suppose that we have a palantir to know what device, controller, game, EmuELEC version, emulator etc etc ... are you using? Please read the wiki first

closed 2021-03-21 by shantigilbert

guy2010, 2021-03-21:
Ipega 9090 joypad 
Emuelec 3.9