EmuELEC/EmuELEC #532: Amlogic processor

dougsky, 2021-03-21:
I have an amlogin s905w X96mini processor ... in the query by the android terminal it mentions that the .dtb to use and the gxl_212_1g, I record the image I set the dtb to start the boot, but the start of the emulec it locks in the introduction video and it is not possible do nothing but turn off the plug ... is there any way to solve it? happens with emuelec version 3.9 and 4.0

[https://youtu.be/79UXrIkSk4U](url)

AlienatedDiego, 2021-03-21:
weird..if gxl\_212\_1g is shown in the terminal I must assume that it is a s905x 1gb ram; in that case the .dtb would be glx\_p212\_1g.dtb

dougsky, 2021-03-21:
[image]
[image]

AlienatedDiego, 2021-03-21:
this would be the first time I see a soc s905w uses a p212.dtb :/ ..try contrasting results using apps like Aida64 or Cpu-z and try all the p212 available in "device trees"

dougsky, 2021-03-21:
aida64 and cpuz return the same hardware information ... right now I am testing other .dtb ..... the vast majority do not even boot

ws02589111, 2021-03-22:
You can try with the below table
```
Device              RAM/STORAGE          CPU          DTB

3GO APLAY 4         2G/16G               S905X        gxl_p212_2g
A95X A1             1G/8G                S905X        gxl_p212_1g
A95X-B7N            2G/8G                S905X        gxl_p212_2g
A95X-R2             1G/8G                S905W        gxl_p281_1g_a95xr2
A95X-R2             2G/16G               S905W        gxl_p281_2g_a95xr2
Abox A1 Max         1G/8G                S905W        gxl_p281_1g_a95xr2
Abox A1 Max         2G/16G               S905W        gxl_p281_2g_a95xr2
Akaso HM8           1G/8G                S905X        gxl_p212_1g
Beelink GT1 Ult.    3G/32G               S912         gxm_q200_3g
Beelink GT1 Ult.    3G/32G(fake RAM)     S912         gxm_q200_2g
Beelink Mini MX     2G/16G               S905         gxbb_p200_2G_1Gbit
Beelink Mini MXIII  2G/16G               S905         gxbb_p200_2G_1Gbit_RealtekWiFi / gxbb_p200_2G_1Gbit
Beelink Mini MXIII II 2G/16G             S905X        gxl_p212_2g    
Bqeel M9C Max       2G/16G               S905X        gxl_p212_2g
Bqeel M9C Pro       1G/8G                S905X        gxl_p212_1g
Docooler M9S Pro    2G/16G(fake RAM)     S905X        gxl_p212_1g_slowemmc
Edal T95Z Plus      2G/16G               S912         gxm_q200_2g
H96 Pro             2G/16G               S912         gxm_q200_2g
H96 Pro             3G/16G               S912         gxm_q200_3g
H96 Pro+            2G/16G               S912         gxm_q201_2g_1gbit
H96 Pro+            3G/16G               S912         gxm_q201_3g_1gbit
KIII                2G/16G               S905         gxbb_p200_2G_1Gbit
Magicsee N4         1G/8G                S905X        gxl_p212_1g
Magicsee N5         2G/16G               S905X        gxl_p212_2g
Mecool BB2 Pro      3G/16G               S912         gxm_q201_3g_1Gbit
Mecool HM8          1G/8G                S905X        gxl_p212_1g
Mecool KI Pro       2G/16G               S905D        gxl_p230_k1_pro
Mecool KII Pro      2G/16G               S905         gxbb_p200_2G_1Gbit
Mecool KIII Pro     3G/16G               S912         gxm_q200_k3_pro
Mecool KM8          2G/16G               S905X        gxl_p212_2g
Mecool KM8 P        1G/8G                S912         gxm_q201_1g
Mecool KM8 Pro      2G/16G               S912         gxm_q201_2g_1Gbit
Mecool M8S Plus     2G/8G                S905         gxbb_p200_2G_1Gbit
Mecool M8S Pro+     2G/16G               S905X        gxl_p212_2g
Mecool M8S Pro L    3G/32G               S912         gxm_q201_3g
Mini M8S 1Gbit      2G/16G               S905         gxbb_p200_2G_1Gbit
Mini M8S            2G/16G               S905         gxbb_p200_2G
Mini M8S II         1G/16G               S905X        gxl_p212_1g
Mini M8S II         2G/16G               S905X        gxl_p212_2g
Minix NEO U1        2G/16G               S905         gxbb_p200_2G_minix_neo_u1
Minix NEO U9-H      2G/16G               S912         gxm_q200_2g_minix_neo_u9
MXQ Pro 4K (old)    2G/8G                S905         gxbb_p201_2G
MXQ Pro 4K (new)    1G/8G                S905X        gxl_p212_1g
Nexbox A1           2G/16G               S912         gxm_q200_2g
SAMMIX R95          2G/16G(fake RAM)     S905X        gxl_p212_1g_slowemmc
Sunvell T95M        2G/8G                S905X        gxl_p212_2g
Sunvell T95N        2G/8G                S905X        gxl_p212_2g
Sunvell T95X        1G/8G                S905X        gxl_p212_1g
Sunvell T95Z Max    2G/16G               S912         gxm_q201_2g_1Gbit
Tanix TX3 Mini      1G/8G                S905W        gxl_p281_1g
Tanix TX3 Mini      2G/16G               S905W        gxl_p281_2g  
Tanix TX3 Pro       1G/8G                S905X        gxl_p212_1g
Tanix TX3 Pro       2G/16G               S905X        gxl_p212_2g
Tanix TX92          3G/32G               S912         gxm_q200_3g
THL Box 1 Pro       2G/16G               S905X        gxl_p212_2g
Tronsmart Vega S95  2G/8G                S905         gxbb_p200_2G_1Gbit
Vontar TX92         3G/32G               S912         gxm_q200_3g
Vorke Z1            3G/32G               S912         gxm_q200_3g
Vorke Z6            3G/32G               S912         gxm_q200_3g
Vorke Z6 Plus       3G/32G               S912         gxm_q201_3g_1gbit
WeTek Play 2        2G/8G                S905         gxbb_p200_2G_wetek_play_2
X92                 3G/32G               S912         gxm_q200_3g / gxm_q201_3g_1gbit
X96                 1G/8G                S905X        gxl_p212_1g
X96                 2G/16G               S905X        gxl_p212_2g
X96                 2G/16G               S905W        gxl_p281_2g
X96 Mini            2G/16G               S905W        gxl_p281_2g
Xtreamer Whizz      2G/16G               S905H        gxbb_p200_2G_1Gbit
Z69                 3G/32G               S905X        gxl_p212_3g
```

dougsky, 2021-03-22:
hello friend, I've tried with all possible .dtb always happens in the same problem, gxl_282_1g and 2g always hangs on the X96MINI logo .... I tested the corelec and presented the same problem ... I believe that this box and false can only .. . mine has 1gb ram

AlienatedDiego, 2021-03-22:
The problem with the x96 is that there are many variations of different manufacturers, different components and of course the cheapest ones, and they are all called the same; in this case it seems that soc, ram or other component are not original and the .dtb that works enters kernel panic..if you have the possibility try to change device for another more reliable one such as beelink, tanix or whatever not "x96"

I have an s905x (p212) that starts the system with the .dtb that includes Emuelec by default (in case you want to try) but it may have other problems with wifi / bt / usb. For devices with 1gb ram better 3.9

dougsky, 2021-03-22:
I have a TX 3 mini ... s905w runs very well, up to version 4.0 ... I bought the X96MINI for the price ... but it wasn't worth it! already left it for use on tv with streaming apps ... thanks a lot for the help !!!

dm2912, 2021-03-22:
> I have a TX 3 mini ... s905w runs very well, up to version 4.0 ... I bought the X96MINI for the price ... but it wasn't worth it! already left it for use on tv with streaming apps ... thanks a lot for the help !!!

if you had a good experience with the Tanix TX3 Mini I would stick with the same brand. I have both the TX3-Mini and the regular TX3 (s905x3) and had no problems with either of them.

dougsky, 2021-03-22:
> O problema com o x96 é que existem muitas variações de fabricantes diferentes, componentes diferentes e, claro, os mais baratos, e todos eles são chamados do mesmo; neste caso, parece que soc, ram ou outro componente não são originais e o .dtb que funciona entra no kernel panic .. se você tiver a possibilidade de trocar o dispositivo por outro mais confiável, como beelink, tanix ou qualquer outro "x96 "
> 
> Eu tenho um s905x (p212) que inicia o sistema com o .dtb que inclui o Emuelec por padrão (caso você queira tentar), mas pode ter outros problemas com wifi / bt / usb. Para dispositivos com 1 gb de RAM melhor 3,9

I'm going to test with the standard dtb ... I haven't done it yet ... I'll give it back as soon as I test it

dougsky, 2021-03-23:
> O problema com o x96 é que existem muitas variações de fabricantes diferentes, componentes diferentes e, claro, os mais baratos, e todos eles são chamados do mesmo; neste caso, parece que soc, ram [...]

tested it with standard dbt ... started the partitioning process ... when it started again ... I had to turn off the box and turn it on again .. when starting emuelec it crashed on the start screen

I give up ... only serves to straming this box

closed 2021-04-12 by shantigilbert