EmuELEC/EmuELEC #533: Some questions for partition

ws02589111, 2021-03-22:
1. I flashed latest dev branch to my card， and now i can't found EEROMS partition on my windows but diskgenius, what should I do to show the EEROMS partition?
In diskgenius, I can see the partition
[image]
[image]

but in windows "computer" i can only found the system partition 
[image]

2.The system partition now is 2gb， this is too large for me， can I set it lower than now like 1gb in my custom build? is this setting has other risk?

thanghn90, 2021-03-24:
For (1): you need to boot to your sd card on your tv box ONCE. Then you'll be able to see the EEROMS partition. The first boot automatically do some initialization like partitioning.
For (2): I would rather make sure the system partition after first boot is not exceeding 1GB before tampering with the dtb file for your tv box. Beside, more emulators/retroarch cores may be added in the futures, so you want to make sure your /system have enough space for that (correct me if I'm wrong).

ws02589111, 2021-03-24:
> For (1): you need to boot to your sd card on your tv box ONCE. Then you'll be able to see the EEROMS partition. The first boot automatically do some initialization like partitioning.
> For (2): I [...]

For(1): This is result of boot to my sd card on my tv box ONCE
For(2): If needed I'd rebuild a image and reinstall my card, I just want to know the lower space has any risk for runtime?（test about 2 days, didn't get any adverse effects now）

thanghn90, 2021-03-24:
(1) That is strange... Anyways, sometimes I got the same issue with Linux sd cards and usb flash drives (armbian's boot partition, specifically). You simply need to assign a driver's letter to that EEROMS partition by going to Start menu, type "Computer Management", then under "Storage" --> "Disk Management", locate your sd card drive and the EEROMS partition (should be the biggest one on the sd card), then "Assign/Change A driver letter path". Then suddenly it works with subsequent sd card image burnings. This is purely WIndows karma. It's the same process as Linux's mounting, but with a lot more button clicks.
(2) How did you resize the system partition? Did you change the dtb or did you resize the partition after boot via ssh command?

ws02589111, 2021-03-24:
> (1) That is strange... Anyways, sometimes I got the same issue with Linux sd cards and usb flash drives (armbian's boot partition, specifically). You simply need to assign a driver's letter to that [...]

1.The method you said is useful，but is there any method to solve the problem by emuelec script or code？
2.Set the `SYSTEM_SIZE=2048` to `SYSTEM_SIZE=1024`  in EmuELEC/distributions/EmuELEC/options file, and the extra 1gb space will add to storage partition

shantigilbert, 2021-04-01:
Assigning the letter to the EEROMS partition needs to be done from the Disk Management in Windows 

But the rest is not an EmuELEC issue, you can set whatever size you like in your custom builds, but updates (and maybe something else) will be broken, that is up to you to fix. 

Closing this as not a real issue.

closed 2021-04-01 by shantigilbert