EmuELEC/EmuELEC #534: Characters (CN, JP, KO) are not displayed properly in RetroArch

ws02589111, 2021-03-23:
**Describe the bug**
This issue get from latest dev branch but the  latest release image  v4.0 is ok
I tried to replace the font in path /tmp/assets/xmb/monochrome/font.ttf and update video_font_path = "/tmp/assets/xmb/monochrome/font.ttf" in retroarch.cfg but no use

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'RetroArch'
2. Change the language to one of CN, JP, KO
3. Save config
4. Restart RetroArch

**Expected behavior**
Display characters normal

**Screenshots**
[image]

**EmuELEC version:**
 -  latest dev branch

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE]
 - SOC: S905M, S905L

ws02589111, 2021-03-23:
I tried revert retroarch to old version to test, until reverted to 7450f049e72dd3388dfed2439c1af73e088e4fd9， the RetroArch  not displayed properly issues is no longer occurs. but the other issues is also existed

shantigilbert, 2021-03-23:
#431

closed 2021-03-23 by shantigilbert

insmagic, 2021-04-28:
not resolved in #431,it worked on 4.0.When I update to 4.1,the retroarch displayed still wrong like this but the game roms are display all right in emuelec.

insmagic, 2021-04-28:
I replace the .ttf files to the storage/.config/retroarch/assets,then change the language to CN,it displayed perfect,but if I quit the retorarch and select restart to retorarch,it displayed all as □□□□

shantigilbert, 2021-04-28:
This is a Retroarch issue,  I could not find a solution so this has to be directed to them.

insmagic, 2021-04-29:
Yes I found that it seems a bug in retroarch 1.9.1,thank you!😄

mavisguo, 2021-06-19:
You know that you can fix it but you don't. It's not the problem of RA. You can't anger all Chinese players just because a Chinese manufacturer wants to pay and ask you to make customized EMUELEC. Please fix it. Please!

shantigilbert, 2021-06-19:
> 
> 
> You know that you can fix it but you don't. It's not the problem of RA. You can't blame all Chinese players just because a Chinese manufacturer wants to pay and ask you to do customized EMUELEC. Please fix it. Please!

if you are so angry, fix it yourself then.

shantigilbert, 2021-06-19:
For future reference: https://github.com/libretro/RetroArch/issues/12215

mavisguo, 2021-06-20:
ok,ok,well.

> > You know that you can fix it but you don't. It's not the problem of RA. You can't blame all Chinese players just because a Chinese manufacturer wants to pay and ask you to do customized EMUELEC. Please fix it. Please!
> 
> If you are so angry, fix it yourself then. I do not know how to fix it, I already told you its a RA issue. Not something I can just flip a switch and fix.

ssmax, 2021-09-24:
in EMUELEC 4.2 , Retroarch version 1.9.4 , 32 bit (don't know why, haha)
when it quit, it will set a hidden config line in retroarch.cfg:
xmb_font = ""
this config will lead to the problem.
you should set it to your font path manually by ssh or samba.
after you set it , you will resolve the problem.
only when you select xmb menu_driver in Emulationsation, it will replace the menu_driver option before start retroarch.