EmuELEC/EmuELEC #538: Install to internal storage?

thanghn90, 2021-03-24:
Any plan to make a script to install EmuELEC directly to the tv box's internal storage, similar to coreelec? I could not google any instruction about this. I'm just asking so that I can use my box's internal storage, not to violate any EmuELEC rules about distribution and licensing...

shantigilbert, 2021-03-24:
how are we supposed to know what device you have?

thanghn90, 2021-03-24:
Well, I thought coreelec installtoemmc / installtointernal / or their new tool called ceemmc is generic rather than box-specific, so I didn't specify my box's model and specs. Anyways, given that you the main dev is asking this question, I bet I myself would need to explore what coreelec install-to-internal script did and try to replicate the same procedure for EmuELEC. Most likely it automatically detect the internal storage size and start partitioning it, then copy the data from SD card to the internal storage. With the new ceemmc tool, it's possible to even dual-boot android and coreelec, as well as backup/restore internal partitions.
Any advice from other geeks who have tried this on EmuELEC is appreciated!

dm2912, 2021-03-24:
> Well, I thought coreelec installtoemmc / installtointernal / or their new tool called ceemmc is generic rather than box-specific,

install to internal has very varying results. Some boxes it works just great, some boxes flat out refuse to either install, or boot to the new partition, and some boxes get bricked completely by the process. 

So what box is very relevant, and a good starting point would be check on the coreelec forums for that box and see if people have been successful in installing to internal. 

That said, in answer to your original question, there is nothing stopping you using the coreelec ceemc process on emuelec, t works just fine ON SUPPORTED BOXES, i installed to internal on both a tx3 mini and a tx3, 

Should note though, on a wide range of devices even coreelec don't support installing to internal because of the associated risks. It is only something you should do if a) you have properly researched your device and its compatability and b) if you are willing to brick the device and/or have android system images for that device ready and know how to reflash it if need be. 

I'm not sure that is something that should be in a built in script where anybody can just click it without any thought for the consequences :D

shantigilbert, 2021-03-24:
@thanghn90 I didn't ask because I don't know, I asked because I do know and @dm2912 explained in detail, installtointernal is include in Amlogic devices and cemmc is included in Amlogic-ng, do understand that doing that is at your own risk and no support other than "reflash android" will be given here nor on the CoreELEC forums. There is still a risk of permanently bricking the device, and not to mention that updates have not been properly tested.

thanghn90, 2021-03-24:
Thanks @shantigilbert and @dm2912 . Looks like nobody is willing to brick their boxes (obviously). Well, I'm glad to be a lab rat then. I bricked my box twice while flashing different roms, and a NAND pin-short + xtmod android image fixed it twice. Right now I'm waiting for my new box to arrive (@dm2912: Tanix TX3 S905X3 4GB/32GB - is this the box you installed emuelec to internal storage?). Once it arrives, I'll test the ceemmc script on my current box and let you know how it goes. Just please don't close this thread yet.
FYI, here is my brick-worthy box: an H96 Max X2 S905X2 Ram 2GB Rom 16GB. Wifi and bluetooth are NOT working even with the stock android image, and Armbian doesn't even recognize the Ethernet port. XTMOD rom doesn't recognize my PS4 controller (but did recognize my PS5 controller, go figure...). So far, CoreELEC and EmuELEC are the only two truly-usable operating systems, but CoreELEC's gamepad management is so inferior to EmuELEC for the moment (no turbo buttons for NES games, no multi-gamepad support).

dm2912, 2021-03-24:
> Tanix TX3 S905X3 4GB/32GB

yup. i had both Emuelec and Coreelec (at separate times) installed in dual boot mode on that box, and currently have coreelec installed in single boot. Have to force it with -x flag, but both times it was fine,

thanghn90, 2021-04-08:
I can confirm ceemmc works for my H96 Max X2 2GB/16GB box. Hell, it's a lot faster to boot emuelec from internal storage! Following @dm2912 advice above, what I did was simply typing:
ceemmc -x
through ssh, then choose option (3): single boot mode, copy from sd card.
So this is basically the same as coreelec's install to internal procedure. More instruction can be found in the official coreelec ceemmc webpage: https://discourse.coreelec.org/t/how-to-use-the-team-coreelec-ceemmc-tool/7630

closed 2021-04-08 by thanghn90