EmuELEC/EmuELEC #539: OpenBor new games not working

Jsinz007, 2021-03-24:
I would like to report an issue with the OpenBor emulator on Emuelec. I am unable to complete the first level with many of my openbor games. After doing some research i have found out that the openbor build currently on Emuelec is old and therefore doesn't allow win animations. Are you able to please update the OpenBor emulator to the newer build like 6412 or above please?

british-choi, 2021-03-29:
@Jsinz007 
Hi~
What is "6412"? this is not commit id.
OpenBor's build in EmuELEC is not that old.

Jsinz007, 2021-03-29:
@british-choi thanks for your reply. No, this is the actual version build used by the Pak creator who has advised me that the Openbor version on Emuelec doesn't seem to support win animations hence why cannot complete first levels. Do you need the commit ID for the version? If so i will ask the games creator.

british-choi, 2021-03-29:
@Jsinz007 
"6412" was the Build Number.
OpenBor's build number on EmuELEC is newer than 6412.
However, in order to solve your issue, you have to make a suggestion to the developer of the OpenBOR engine.

Jsinz007, 2021-03-29:
Ok what would the Openbor developers need to do?

british-choi, 2021-03-29:
@Jsinz007 
The link below is the OpenBOR engine developer's github.
Register your issue
https://github.com/DCurrent/openbor

Jsinz007, 2021-03-29:
I am not sure what i am asking them to do? The Openbor engine works on my pc and other devices. My issue is it simply doesn't work on Emuelec.

shantigilbert, 2021-03-29:
@Jsinz007 neither do we... If the build is newer than what you need and you still have issues, you most likely need to ask upstream, I only compile the engine, I do not make changes to it.

shantigilbert, 2021-04-01:
I will close this, feel free to reopen with new information. 

You did not include the EmuELEC version you are using, but here is more info for you.

v4.0 uses: 2b7f9ac6cf244e9e68da5e85459e2ba9c5bcb0b9
Latest 4.1 test uses: 60b7e9687ed1d3682b84067c9f47d2ffe1f8cab9

Both of which are ahead of the required "6412" I assume since "7142" was on on Dec 8, 2019 and v4.0 hash is from Jan 2021

closed 2021-04-01 by shantigilbert