EmuELEC/EmuELEC #541: Emulec Super Console X Pro

thugmp69, 2021-03-26:
I have a Super console X pro that runs Emuelec. I wanted to change the resolution from its original state in emuelec settings and i chose the option auto detect resolution. I had to reboot the device and i did so...after the device started again i saw the Emuelec logo and after that i got a solid green screen. Now im not able to see any kind of menus and and can't switch the resolution as i don't see anything. I read about an issue similar to mine but the ways that were suggested there would not work for me. I would really appreciate if someone had a similar issue and can help me fix it.

dm2912, 2021-03-26:
SuperConsole X is not a supported device unfortunately. You will need to get in touch with the manufacturer or supplier. 

https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices

ws02589111, 2021-03-26:
#536

thugmp69, 2021-03-26:
I see...thanks anyway guys!

closed 2021-03-26 by shantigilbert