EmuELEC/EmuELEC #542: 4.0 and 4.1, access to multiple controllers at the same time, running 32-bit PSX, controller sequence error

zcl226403, 2021-03-26:
My English is not good, use the translation software, very sorry
1.
4.0 and 4.1, access to multiple controllers at the same time, running 32-bit PSX, controller sequence error
Running the 64-bit core, the controller and the controller displayed on the front end are in normal order
2.Version 4.1, in 64-bit Retroarch, garbled after setting Chinese, but normal in 32-bit Retroarch

zcl226403, 2021-03-31:
1.PSX system, controller sequence error
2.64 bit retroarch, Chinese garbled code

shantigilbert, 2021-04-01:
I do not understand what you are trying to say.

Also, PLEASE follow the template! I will close this issue, and feel free to reopen with the correct template, and a better example.

closed 2021-04-01 by shantigilbert