EmuELEC/EmuELEC #543: Screenscraper API closed for non-registered members

tanathos, 2021-03-26:
It seems the username/password for screenscraper are not correctly used while connecting to the API: I keep receiving the message "api closed for non-registered members" even if I'm a registered member.
In fact this is happening anytime on the site ([https://www.screenscraper.fr/](https://www.screenscraper.fr/)) they're stating that the server is busy.

I'm using Emuelec 4.0 on a ODROID GO Super.

shantigilbert, 2021-03-26:
> they're stating that the server is busy.

Try again later, the error is clear and it has nothing to do with EmuELEC

Closing this issue, feel free to reopen if its in fact an EmuELEC problem.

closed 2021-03-26 by shantigilbert

tanathos, 2021-03-26:
What I mean is: even if the server is busy for registered users the usage of the API should always be allowed, so my suspect here is that the call to the API is done as anonymous user.
They have the sad face near not registered, but registered user should still use it:
[image]
The hover message of the face is "Due to high CPU consumption, the API is currently closed to non-registered users".

shantigilbert, 2021-03-26:
Please check your username and password and make sure it is working on the ss website, as I cannot reproduce your issue. I can scrape with no problems (just tested it)

tanathos, 2021-03-27:
Yeah, the credential are working on the website, and I rechecked them more than once.