EmuELEC/EmuELEC #544: Wifi adapter Edimax EW-7811Un V2 not work

daydaychao, 2021-03-28:
Wifi adapter Edimax EW-7811Un V2 not work.

I set SSID and Password so many times but it no networks found.
Please check it out

https://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/us/wireless_adapters_n150/ew-7811un_v2/

dm2912, 2021-03-28:
Can you find out what chip is actually in it.

Because strangely, nowhere on their site, or in any of the technical details do they say what chip is in it. 

Looking at amazon q&a seems a lot of people have issues running this dongle on linux https://www.amazon.co.uk/ask/questions/Tx1A4DYWE7NQVXN/ref=ask_dp_dpmw_al_hza and the device was not supported as of kernel 4.13

miwasp, 2021-03-28:
There are 2 versions of this stick.
Version 1 Edimax EW-7811un USB Ids 7932:7811 has a RTL8192cu chipset and works perfectly.
Version 2 Edimax EW-7811un USB Ids 7932:b811 uses a different RealTek chipset RTL8188EU which needs a different driver that is not included.
Optically both are identical.

daydaychao, 2021-03-29:
Thank you dm2912, miwasp.

closed 2021-03-29 by daydaychao