EmuELEC/EmuELEC #546: Incorrect time zone entry `Asia/Hong_ong` which should be `Asia/Hong_Kong`

PikachuEXE, 2021-03-30:
**Describe the bug**
A clear and concise description of what the bug is.
Odroid-GO Super has displayed a time zone entry value `Asia/Hong_ong` in time zone config menu

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Main Menu', 'System Setting', 'Timezone'
2. Scroll down to 'Asia/HOVD'
3. Look at the entry above 'Asia/HOVD'

**Expected behavior**
1. Displayed as `ASIA/HONG_KONG`
2. When it's selected as timezone, system clock display time in UTC+8

**Actual behavior**
1. Displayed as `ASIA/HONG_ONG`
2. When it's selected as timezone, system clock display time in UTC+0

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
v4.0
(Has not tried on latest test version yet)

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
Odroid-GO Super

british-choi, 2021-03-31:
@shantigilbert 
emuelec-emulationstation
diff --git a/es-core/src/platform.cpp b/es-core/src/platform.cpp
index 82d4704b..655e53c0 100644
--- a/es-core/src/platform.cpp
+++ b/es-core/src/platform.cpp
@@ -381,7 +381,7 @@ std::string getShOutput(const std::string& mStr)
     while(fgets(buffer, sizeof(buffer), pipe) != NULL)
     {
         file = buffer;
-        result += file.substr(0, file.size() - 1);
+        result += file.substr(0, file.size());
     }

     pclose(pipe);

shantigilbert, 2021-03-31:
> 
> 
> @shantigilbert
> 
> emuelec-emulationstation / platform.cpp
> check below function
> std::string getShOutput(const std::string& mStr)
> `- result += file.substr(0, file.size() - 1);`
> `+ result += file.substr(0, file.size());`

does this fixes the issue? does it cause any other issues? `getShOutput` is used on many places

TheLastProject, 2021-03-31:
It's interesting that this only seems to happen on the Odroid Go Super. My GameForce Chi is not affected. May be a version difference though, my OGS is on V4.1-TEST-02182021 and my GFC is on V4.1-TEST-03082021.

I thought maybe the zone1970.tab file was different (see https://github.com/EmuELEC/EmuELEC/pull/279/files#diff-c903f109b5cb8349fe96edbe90f34eabe6a13627e6877d4611e97794b37e27a5R193) but it's not, it's written as "Asia/Hong_Kong" there correctly.

shantigilbert, 2021-03-31:
I think its even more strange that it only happens with that particular string none of the others seem affected, and none of the other places that use the same function are affected either... I am thinking there is something that changes `_K` to `_` somewhere in the code but I have not looked into it yet.

PikachuEXE, 2021-03-31:
My `/usr/share/zoneinfo/zone1970.tab` on device does have `Asia/Hong_Kong`

Another incorrect timezone spotted would be `Amrica/Nipigon` which should be `America/Nipigon` (hard to find it in file)

british-choi, 2021-04-01:
@shantigilbert 
this fixes the issue!
string substr(size_t pos = 0, size_t len = npos) const;
len parameter should be file size(max 256 byte) in reading pipe.
However, 256 bytes were read from the pipe, and only 255 bytes were copied to the result.
So some characters are missing.

shantigilbert, 2021-04-01:
I just tested that and it causes a LOT more issues, 
Shaders no longer work for example. 

But this gives an idea as to why there are broken, so I will investigate 

I think that `/usr/bin/emuelec-utils timezones` should return a list of TZ with a new line at the end for it to work

shantigilbert, 2021-04-01:
```
function timezones() {
	cat /usr/share/zoneinfo/zone1970.tab | grep -v "^#" | awk '{ print $3"," }' | sort -u
}
```
in https://github.com/EmuELEC/EmuELEC/blob/f5d53ae824b9e721b9d128ac74236f648d9699b9/packages/sx05re/emuelec/bin/emuelec-utils#L171

This is the fix, ES code is untouched and now both of those TZ display correctly

shantigilbert, 2021-04-01:
[image]

PikachuEXE, 2021-04-01:
Will try again once next test version released

commit ae297d47 (2021-04-01): emuelec-utils: Fix some TimeZones not displaying/working  correctly 

Fixes: #546
closed 2021-04-24 by shantigilbert