EmuELEC/EmuELEC #550: Emuelec 4.0 screen issue with x96 max+ with s905x3 4g

eranlenga, 2021-04-01:
hi.
i have x96 max+ with s905x3 4gb ram 64gb storage .
i'v made sd card with Emuelec 4.0.
when i turn on the tv box with the sd card for the first time i see Emuelec logo. 
after that i see the Emuelec start up screen but in a wired way. the picture multiply 3 time.
every version of Emuelec doesnt work good. i tried with 4k TV and 720P TV with HDMI cable
​​​​​​​any idea?

[image]
[image]

shantigilbert, 2021-04-02:
Set your Android TV box to 1080p, then reboot to EmuELEC

closed 2021-04-02 by shantigilbert

eranlenga, 2021-04-02:
I did it already. 
Maybe it something i can change in the config.ini file?