EmuELEC/EmuELEC #552: BezelProject GBA and Inability to change core aspect ratio

aqualime-bit, 2021-04-04:
**Describe the bug**
Bezelproject ratio is wrong for GBA and cannot adjust the core aspect ratio in the Emuelec menu.

**To Reproduce**
Steps to reproduce the behavior:
1. Load games into GBA
2. Play game to verify correct aspect ratio
3. Install BezelProject for GBA
4.Enable RA Bezels 
5. Load same game again

**Expected behavior**
The game will now have a much much smaller aspect ratio. Changing the options via "Game Settings, Per System advanced Config," Will have zero affect. Changing the ratio inside Retroarch back to "Core Provided" fixes the issue, but won't save... or I don't know how to save it.

**EmuELEC version:*4.0*
 - Version [e.g. 22]

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: S095x3
 - Brand: T95
 - Model Max Plus

shantigilbert, 2021-04-11:
Since the bezel project changed the bezels, the script is still waiting for the old ones, if you are in v4.0 edit `bezels.sh` and change to your liking, else you need to wait for a new version and someone what has tested and has the correct settings.

shantigilbert, 2021-04-12:
Can you please check if these other handhelds are doing the same thing? 

"gamegear", "gb", "gbc", "ngp", "ngpc", "wonderswan", "wonderswancolor"

AlienatedDiego, 2021-04-12:
To save custom aspect ratio changes go to Settings / user interface / views / quick menu / save core overrides and save game overrides (ON). Once activated they appear in quick menu / overrides

AlienatedDiego, 2021-04-12:
[image]

AlienatedDiego, 2021-04-12:
> ¿Puede comprobar si estas otras computadoras de mano están haciendo lo mismo?
> 
> "gamegear", "gb", "gbc", "ngp", "ngpc", "wonderswan", "wonderswancolor"

Yes, this happens with bezel downloaded from "The Bezel Project" on all portable systems (wonderswan ones untested)

closed 2021-04-24 by shantigilbert