EmuELEC/EmuELEC #553: 4.1-TEST-03302021 / OGA 1.0 / FBNeo Vertical mode: up/down controls are backwards.

cplr, 2021-04-05:
Using "vertical mode" sort of works on OGA 1.0, except the controls are not "rotated" properly. Left and Right work, however pressing Up (which technically is DPad-Right, but rotated) really sends Down and pressing Down (which technically is DPad-Left, but rotated) really sends Up.

**To Reproduce**
Steps to reproduce the behavior:
1. Add 1941 to fbneo directory
2. Configure EmuELEC to use Vertical mode for the game
3. Start 1941
4. Play for a bit. Left and Right are correct. Pressing up while rotated sends down. Down sends up.

**Expected behavior**
For up and down to be properly rotated as well.

**EmuELEC version:**
- Version 4.1-TEST-03302021-odroidgo2

**Device**
Odroid Go Advance v1.0

shantigilbert, 2021-04-05:
I unfortunately do not have a v1.0 anymore so I cannot test this, but in theory it should be easy to change this file: https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec/gamepads/GO-Advance%20Gamepad_vertical.cfg to the corresponding values.

british-choi, 2021-04-08:
https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec/gamepads/GO-Advance%20Gamepad_vertical.cfg 
input_up_btn = "8"  =>  input_up_btn = "9"
input_down_btn = "9" => input_down_btn = "8"

shantigilbert, 2021-04-08:
> https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec/gamepads/GO-Advance%20Gamepad_vertical.cfg
> input_up_btn = "8" => input_up_btn = "9"
> input_down_btn = "9" => input_down_btn = "8"

Yeah makes sense, thanks!

closed 2021-04-24 by shantigilbert