EmuELEC/EmuELEC #558: tried everything, but I can't get emuelec to work

mook4, 2021-04-11:
Hi!
I bought an new X96 Max Plus 4TB/23GB Amologic S905x3 tv box (Geekbuying), especially to play games with emuelec. 

Installation just works without errors, menu also works, sound works, game controller works. 
However, when I start a game (e.g. Doom, an already installed game) I get a red screen (see photo)
[image]

when I put the USB stick or SD card in my Windows 10 computer, I don't get a Fat 32 (EEROMS) partition. (see picture)
[image]

If I do the " check Missing Bios"  in the Game Options, I get many error messages.

specifications of my box: (by aida64 and terminal emulator apps)
[image]
[image]
[image]
[image]

I used this file:  EmuELEC-Amlogic-ng.aarch64-4.0-Generic.img.gz
I used Etcher and Rufus to burn this file to a USB or SD (tried both, same result)
I tried a couple of device trees, but this one should be it (seems to me): sm1_s905x3_4g_1gbit.dtb

hopefully somebody can help me! Greetings from the Netherlands

shantigilbert, 2021-04-11:
Emuelec does not occluded any games, those are launchers you are responsible for adding the files required.

mook4, 2021-04-11:
Thank you for your quick response!
 
But if I want to add Roms:
when I put the USB stick or SD card in my Windows 10 computer, I don't get a Fat 32 (EEROMS) partition. (see picture)

mook4, 2021-04-11:
If I do the " check Missing Bios" in the Game Options, I get hundreds error messages
[image]

mook4, 2021-04-11:
Like to hear from you.

shantigilbert, 2021-04-11:
@mook4 no idea, you need to wait until the partition is created on first boot, if not then it's a bad flash most likely.

As for the bios, doesn't it seem obvious?? The option is called "missing bios" it's showing you what bios are missing.

Since this is not really a big, but user error, please use discord for this kind of things.

closed 2021-04-11 by shantigilbert

mook4, 2021-04-11:
Hi,
sorry for the inconvenience
i thought the missing bios was important to emuelec. thanks to your tip that games are not included, I can now manage via the network.
SNES games works now after a lot of days trying  😀

if anyone has a tip on how to get the Fat 32 (EEROMS) partition working please let me know. I tried different SD en USB cards. Thank you

lethal-guitar, 2021-04-11:
@mook4

> if anyone has a tip on how to get the Fat 32 (EEROMS) partition working please let me know. I tried different SD en USB cards. Thank you

You might need to go into disk management and assign a drive letter for the partition. It happens sometimes that Windows doesn't do it automatically. You should be able to find instructions via Google. If you get stuck, feel free to ask for help on Discord (you can find the link in the README/on the Wiki). But please don't ask in this GitHub issue, GitHub issues are only for bugs

shantigilbert, 2021-04-11:
> i thought the missing bios was important to emuelec.

It is, as if an emulator needs them it won't work, but I mean the "errors" are expected if you do not have them

AlienatedDiego, 2021-04-12:
> if anyone has a tip on how to get the Fat 32 (EEROMS) partition working please let me know. I tried different SD en USB cards. Thank you

AlienatedDiego, 2021-04-12:
> if anyone has a tip on how to get the Fat 32 (EEROMS) partition working please let me know. I tried different SD en USB cards. 

Español
Inglés
En ocasiones ,como ya han comentado sólo debes asignar letra a la partición en el administrador de discos de windows,algunas versiones de Windows (en mi caso win7x86) no detectan particiones en dispositivos extraibles(sd,mc,usb).si estás muy interesado existen por ahí fix para que sean detectados como hdd internos .pero es más comodo usando red compartida realmente
Sometimes, as they have already commented, you only have to assign a letter to the partition in the windows disk manager, some versions of Windows (in my case win7x86) do not detect partitions on removable devices (sd, mc, usb). If you are very interested there are fixes out there so that they are detected as internal hdd. but it is more comfortable using shared network really