EmuELEC/EmuELEC #560: ES-front will freeze after return to the previous menu

tryLater, 2021-04-13:
**Describe the bug**
Hello, I upgraded the system from 3.8 to 4.1 beta recently, and I found the ES-front's menu have performance issues
In ES menu， when i change some settings then return to previous menu， the menu will freeze some seconds(>10s)

**To Reproduce**
Steps to reproduce the behavior:
1. Press start
2. Change some setting whatever
3. Back to main menu

**Expected behavior**
The menu back to previous is more quick like 3.8， it's too slow now

**Screenshots**
These are some clues I have observed， when the menu freeze occuring， the emulationstatio occupies 100% of the CPU until the menu freeze disappears
[image]

**EmuELEC version:**
 - Version 4.1 beta with dev branch， commit:  #8621ab4aa65083aa4e625c55f8711d816e929120

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: S905L

AlienatedDiego, 2021-04-13:
I must assume that your device has 1gb ram, this is not enough to manage 4.0+ without problems, minimum requires 2gb ram

The recommended thing for 1gb ram devices is to stay in the latest 32bit version (3.9)

If you decide to stay with 4.x despite the possible problems, use a lighter theme that consumes fewer resources

dm2912, 2021-04-13:
you can try increasing video ram

system settings > developer > and hten up the vram limit to 200mb or so but obviously if your device only has 1gb ram then this is a fair percentage of that

the biggest drain on low memory devices is the number of roms you have. If you only have a few hundred roms, the system will not eat much ram. If you have many thousands, then the system tries to cache information on all of them whenever you enter a gamelist.

Basically, if the device only has 1gb ram, then keep your roms to a sensible level. If you have 1000's upon 1000's of roms you will get memory issues.

tryLater, 2021-04-18:
Thanks for reply guys.
Yes, my device has 1gb ram only，and I try to set vram limit to 200mb and delete some games what I don't like， but it's no obvious effect. now I downgrade my device to my old 3.8 system. maybe I really need to buy a new 2gb device :)

closed 2021-04-18 by tryLater