EmuELEC/EmuELEC #561: Add Support for new variant, Amlogic S905W p282 , the p281 don't boot properly. i tried all

KirbDev, 2021-04-13:
Like i said before, i bought unafortunately some tv box with the variant P282 , equivalent to S905W with 2gb ram and 16 rom specs. the device in question is T95 S1 , i tried executing emuelec for P281 1gb, boots but it turn his screen in black.

How is possible other tv boxes can run emuelec and this model can't?

(sorry broken grammar)

Context: i attach some pictures for better understanding
[image]
[image]
[image]
[image]

dm2912, 2021-04-13:
the trouble with all these generic t95 / h96 / x88 / etc etc is they can all have very different internal components so frequently they will run into dtb errors because the hardware does not exactly match what the dtb expects. 

people have struggled to get coreelec running on this box as well https://discourse.coreelec.org/t/t95-s1-which-device-tree/12058/18

KirbDev, 2021-04-14:
oh i see, for now coreelec and emuelec are currently unsupported for my device?

KirbDev, 2021-04-14:
i think, no ... basically i gonna close the thread. sadly i think gonna use slow emmc boot up

KirbDev, 2021-04-14:
Solved i do the test with some help, i used EE Video mode ntsc with p281 dtb device tree (1G)

closed 2021-04-14 by KirbDev