EmuELEC/EmuELEC #564: Deleting all game metadata from game in collection crashes ES

chucksmith, 2021-04-15:
**Describe the bug**
When I delete a game's metadata from a collection, ES crashes and restarts.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any game within a collection
2. Click on EDIT THIS GAME'S METADATA
3. Click on DELETE
4. Confirm the deletion
5. ES crashes and restarts

**Expected behavior**
I expect it to delete that game from my drive. (I usually do this when I discover duplicate games in my library while browsing collections.)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - 4.1-TEST-03302021

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: S922x
 - Brand: Beelink
 - Model: GT-King

**Additional context**
Add any other context about the problem here.

shantigilbert, 2021-04-24:
Thanks for the report, I cannot reproduce it on newer test so maybe its already fixed, but just in case, does this happen on ALL systems? or just some ?

chucksmith, 2021-04-26:
It doesn't happen on any systems, only on custom collections like the list of 2-player games.

chucksmith, 2021-05-18:
This is no longer a bug in EmuELEC 4.2-TEST-05142021. Feel free to close this issue!

closed 2021-05-18 by chucksmith