EmuELEC/EmuELEC #565: Hello, I got TX3 Mini, i am unable to determine the verison and dtb file, i read the guide and made several trials with no luck, can anyone help?

amhayawan, 2021-04-15:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

dm2912, 2021-04-15:
The DTB for this device (s905w) is gxl_p281_1g

However, there have been numerous reports of fake Mini's in the last month or so with different internal components, some even have had rockchip and allwinner cpu's in them.

This is a genuine mini 

[image]

this is one of the fake/clone boxes. You will notice a) the front display is not at the front of the box, and it has an additional led. 

[image]

first step with this box is trying that specific dtb. If that doesnt work establish if the box actually has an amlogic chip.

amhayawan, 2021-04-15:
Hello,

Thank you for your response.
I am still confused, Im sorry for being so much trouble.

First I downloaded the v4 EmuELEC-Amlogic-ng.aarch64-4.0-Generic. Is this
even the correct version. Second I in the dtb file list I cannot find the
gxl_p281 file.

amhayawan, 2021-04-15:
Hello Again,

I checked using cpu-z: Processor is 8x ARM ***@***.*** ghz.

AlienatedDiego, 2021-04-15:
Wrong version, you need the (non-NG) considering your device is original s905w (glx_p281_1g.dtb). You should check which amlogic processor your device has first.

dm2912, 2021-04-15:
Personally on this box which is fairly underpowered I would use the non-ng version of 3.9

https://github.com/EmuELEC/EmuELEC/releases/download/v3.9/EmuELEC-Amlogic.arm-3.9-Generic.img.gz

[image]

amhayawan, 2021-04-15:
I finally managed to reach the emuelec boot screen, but its stuck.

dm2912, 2021-04-15:
that almost certainly means the install failed which usually means some sort of SD issue, i would reflash the SD and start again.

Before doing that worth checking your sd and making sure it is genuine using something like h2testw

flash using something like Etcher that does a verification test afterwards

second step make sure you use the right dtb, 1g v 2g depending on your device

And this is a genuine mini not a clone as detailed in my first post?

amhayawan, 2021-04-15:
Ill try another sd card, however not sure if its a clone or genuine.

closed 2021-04-24 by shantigilbert