EmuELEC/EmuELEC #566: [Request] PSX Emulation to DuckStation

imadeput0287, 2021-04-22:
Currently we only have PCSX-ReARMed for PSX Emulation, can we get DuckStation (or SwanStation on RA)? It speed pretty good and we can try to upscale to 4x or more (only 2x on PCSX-ReARMed).

Thank you.

dm2912, 2021-04-22:
Its already there in 4.1

My experience so far is its only really viable with s922 chips, I didnt get great performance on s905x3

dm2912, 2021-04-22:
I would be absolutely amazed if it works well on the S905W to be honest.

Its ok on S922, but even my S905x3 works better with Re-Armed and that is waaaay more powerful than a "w"

shantigilbert, 2021-04-22:
Duckstation is already included, and Duckstation is awesome, just not on ARM/aarch64 just yet, specially with MALI, so for the moment there is no way to make such low powered device be the powerhouse you think it is.

closed 2021-04-22 by shantigilbert