EmuELEC/EmuELEC #570: NEC PC-FX music issues

Cebyo, 2021-04-24:
Hi the team.

First of all, I would like to thank you for the wonderful work you do.

I'm writing to you because I am having problems with the emulation of the pc-fx which is a console that I love. The problem is that the sound lags. Do you have any advice for me.

For information, I am running emuELEC version 3.9 on an H96 Max X3 android box. The games I use are in several bin and one cue formats.

Thank you in advance

shantigilbert, 2021-04-24:
No idea, you need to use v4.0 and see if the same issue happens there, other wise your device might not be powerful enough to emulate it

closed 2021-04-24 by shantigilbert

Cebyo, 2021-04-26:
Hello,

this weekend I updated my system to 4.0 but the problem persists. I did indeed think about the power of the Android Box but on the other hand, when I see that the Dreamcast or PSP games work very well, i do not understand how this technically less powerful console has trouble working.