EmuELEC/EmuELEC #572: Two timezone menus in system settings

british-choi, 2021-04-25:
**Describe the bug**
There are two time zone menus.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'system settings' menu
2. Check timezone menu (information / TIMEZONE / language / TIMEZONE)

**Expected behavior**
There should be only one time zone menu.

**EmuELEC version:**
 - 4.1

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: rk3326
 - Brand: hardkernel
 - Model: oga v1.1

PikachuEXE, 2021-05-07:
Same for 
EmuELEC: 4.1
SOC: RK3326
Brand: HardKernel
Model: OGS

No issue in 4.0

PikachuEXE, 2021-05-16:
Don't see it in `4.2-TEST-05142021`
what changed?

closed 2021-06-18 by shantigilbert