EmuELEC/EmuELEC #573: Clock is not shown in emulationstation (OGA v1.1)

british-choi, 2021-04-25:
**Describe the bug**
clock is not shown in emulationstation

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'UI Settings'
2. SHOW CLOCK is ON
3. Back to home
4. Not found current clock

**Expected behavior**
Clock should be shown

**EmuELEC version:**
 - Version 4.1

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: rk3326
 - Brand: hardkernel
 - Model: OGA v1.1

shantigilbert, 2021-04-25:
Cannot reproduce, clock is shown on my OGS. Are you using the default theme?

miwasp, 2021-04-25:
The clock is being displayed on OGS but not on OGA, both with 4.1 and Crystal theme.

british-choi, 2021-04-26:
> Cannot reproduce, clock is shown on my OGS. Are you using the default theme?

I used Crystal theme. OGS is ok, but OGA is NG
Maybe the clock component seems to be positioned too far to the right.

shantigilbert, 2021-04-26:
Got it, will look into it. What happens if you use any other theme?

british-choi, 2021-04-26:
> Got it, will look into it. What happens if you use any other theme?

I used EmuELEC-carbon.
that is normal when set to 24h display.
However, if it is set to 12h display, it will be displayed as "12:30...". (AM/PM is not displayed)

shantigilbert, 2021-04-26:
Ok, but that is another issue altogether. Just want to know if the clock is displayed in other themes

british-choi, 2021-04-26:
> Ok, but that is another issue altogether. Just want to know if the clock is displayed in other themes

I only tested 2 themes. (Crystal, Carbon)
The clock is not displayed in the Crystal theme.
In Carbon theme, a clock is displayed.

dm2912, 2021-04-26:
yeh, ive done nothing to crystal to hide or move the clock, i need to look at carbon to see what he does to have it showing.

Carbon is a really bizarre theme though as far as file structure goes so might take me a bit lol

dm2912, 2021-04-26:
ok, having look, Alekfull solo has clock and thats a far easier theme to look at, so this is something i will address in crystal.

If you want to raise issue on crystal's git feel free so i dont forget and in case others have same issue.

dm2912, 2021-04-26:
Fixed in latest Crystal

no idea what was going on there, i never actually defined or set any criteria for textname=clock so it should have just used ES defaults. 

but ive force defined now so should appear.

british-choi, 2021-04-26:
I have found the code that is causing it.

emulationstation
`es-core/src/Window.cpp`

before code (not displayed clock)
`               mClock->setPosition(Renderer::getScreenWidth()*0.94, Renderer::getScreenHeight()*0.9965 - Font::get(FONT_SIZE_SMALL)->getHeight());`
`		mClock->setSize(Renderer::getScreenWidth()*0.05, 0);`

after code (displayed clock)
`		mClock->setPosition(Renderer::getScreenWidth()*0.90, Renderer::getScreenHeight()*0.9965 - Font::get(FONT_SIZE_SMALL)->getHeight());`
`		mClock->setSize(Renderer::getScreenWidth()*0.10, 0);`

miwasp, 2021-04-26:
With the latest update of the crystal theme the clock is being displayed on both OGS and OGA. Fixed.

dm2912, 2021-04-26:
yeh, im pretty sure this is initially an ES issue as I had not made any changes at all to the clock config in Crystal and @british-choi 's edits to Window.cpp seem to point that way

But, i noticed that in Carbon the clock position is forcibly defined in the main theme.xml so by doing that it now uses the theme positioning, rather than the default ES positioning. That said, I am using trickery to adapt to 4:3 / 3:2 displays so its possible in doing that i caused it to break itself :'). Probably best practice for these things to be handled by the theme anyway.

british-choi, 2021-04-26:
> yeh, im pretty sure this is initially an ES issue as I had not made any changes at all to the clock config in Crystal and @british-choi 's edits to Window.cpp seem to point that way
> 
> But, i noticed that in Carbon the clock position is forcibly defined in the main theme.xml so by doing that it now uses the theme positioning, rather than the default ES positioning. Probably best practice for these things to be handled by the theme anyway.

@dm2912 
I agree with your opinion.
thanks

shantigilbert, 2021-04-27:
> I have found the code that is causing it.
> 
> emulationstation
> `es-core/src/Window.cpp`
> 
> before code (not displayed clock)
> ` mClock->setPosition(Renderer::getScreenWidth()*0.94, Renderer::getScreenHeight()*0.9965 - Font::get(FONT_SIZE_SMALL)->getHeight());`
> ` mClock->setSize(Renderer::getScreenWidth()*0.05, 0);`
> 
> after code (displayed clock)
> ` mClock->setPosition(Renderer::getScreenWidth()*0.89, Renderer::getScreenHeight()*0.9965 - Font::get(FONT_SIZE_SMALL)->getHeight());`
> ` mClock->setSize(Renderer::getScreenWidth()*0.10, 0);`

Why not open a PR on the EmuELEC-Emulationstation github?

british-choi, 2021-04-27:
> Why not open a PR on the EmuELEC-Emulationstation github?

I have never used PR in github.
I'll try.

shantigilbert, 2021-04-27:
What is weird is that part of the code has not changed since 2019... so I wonder what else changed to make this issue suddenly appear. 

to open a PR follow this guide: https://github.com/EmuELEC/EmuELEC/wiki/Developers-corner---How-to-submit-a-pull-request-(PR)

british-choi, 2021-04-27:
@shantigilbert 
In my opinion, rather than modifying the code of the emulationstation, the crystal theme needs the clock area (already modified by dm2912)
The emulationstation code has not been modified, but the clock font has been changed to a wide font.
So, the horizontal size of the clock component should be changed according to the font. I think it is correct that this is applied in the theme.

dm2912, 2021-04-27:
I would agree that best practice is for this to be addressed in the theme. 

Basically if clock is not defined in the theme.xml then the underlying code of ES is broken and the clock does not appear on small screen devices

But as soon as its defined in the theme everything is fine. 

For future reference if other themes are affected by it the relevant code that can go in the <screen> section is

`  <view name="screen">
  <text name="clock">
      <color>777777FF</color>
      <fontSize>0.035</fontSize>
      <pos>0.90 0.94</pos>
      <size>0.09 0.058</size>
      <alignment>right</alignment>
      <verticalAlignment>center</verticalAlignment>
    </text>
   </view>`

shantigilbert, 2021-04-27:
> that said, it wouldnt hurt to fix the underlying code to stop it even being an issue.

dm2912, 2021-04-27:
certainly it was present on the gameforce and was fixed by me forcing position in crystal

closed 2021-06-18 by shantigilbert