EmuELEC/EmuELEC #577: Known Issues v4.1

shantigilbert, 2021-04-25:
### Fixed in Dev:    
- [x] Brigness is not saved correctly on boot (OGA/OGS/GF) (#574)

### Not yet fixed

**Amlogic-ng**
- [ ] Dolphin needs automatic gamepad configuration (#576)

**Odroid Go Advance/Super**
- [ ] Kernel panic (or other random issues, like waking up from sleep) maybe caused by some WiFi dongles (Realtek RTL8188CUS)
- [ ] Clock is not shown in emulationstation (#573)
- [ ] Two timezone menus in system settings (#572)

bennington111, 2021-04-27:
Black screen if use Swanstation core on x96 mini (s905w) 2/16. Sound is ok, but black screen.

PikachuEXE, 2021-05-07:
https://github.com/EmuELEC/EmuELEC/issues/589
This issue is fixed in dev, but not tested yet

iCanuck, 2021-05-08:
Do you have rough idea when dolphin auto gamepad config will be added? Thanks for all your work.

leonkiss, 2021-06-03:
start 4.1  Display the overscan screen ，can't do anything. how to fix it?

SamuEDL, 2021-06-03:
> 
> 
> start 4.1 Display the overscan screen ，can't do anything. how to fix it?

Some televisions allow to deactivate the "overscan". Check if possible in yours.

closed 2021-06-07 by shantigilbert