EmuELEC/EmuELEC #578: Blargg NTSC filter Genesis Plus GX issue

madmax2069, 2021-04-27:
**Describe the bug**
When you try to enable the Blargg NTSC filter in Retroarch for the Genesis plus GX core the screen freezes at Retroarch option menu but you can still hear the game sounds playing when you press the shortcut button to return to game.

**To Reproduce**
Load up any Sega Genesis game using the Genesis plus GX core, go into options and enable any of the options for the Blargg NTSC filter

**Expected behavior**
Run a genesis game with the filter applied with out freezing the screen

**EmuELEC version:**
 - Version 4.0 and 4.1

**Device**
 - SOC: S905x3
 - Brand: don't know
 - Model X88 Pro mini
 - Version 4g/32g

**Additional context**
The Blargg NTSC filter works just fine for SNES in Retroarch in EmueELEC 4.0 and 4.1, but not for Genesis. I figured it was just an issue with either Retroarch and or the core, so I downloaded the standalone version of Retroarch for Android to test to see if the issue is in the stand alone version as well, but the Blargg NTSC filter works just fine for genesis for the standalone Android version. so it would seem its only an issue with Retroarch and Genesis plus GX running EmuELEC.

shantigilbert, 2021-04-27:
> I figured it was just an issue with either Retroarch and or the core, so I downloaded the standalone version of Retroarch for Android to test to see if the issue is in the stand alone version as [...]

You need to test the exact same version of the git core, this might be an issue not present in one version but present in the other, if it works in other emulators then its obvious the issue is with the GX core, but for this to be fixed it needs to be tested in other versions as well.

madmax2069, 2021-04-28:
so, what do I need to do then.

because all I know is that the Blargg filter in GX in the stand alone android version of retroarch works fine, and that the whatever version of retroarch and GX that's included with emuelec (both 4.0 and 4.1) the Blargg filter doesn't work.

shantigilbert, 2021-06-07:
I think I found the issue and I will try to fix it ASAP.
To test this, when you get the issue, connect by SSH and type this `fbfix` and see if the image returns.

commit 818cddf3 (2021-06-07): Amlogic-ng: Fix Retroarch not updating after applying CPU video filter

Fixes #578
closed 2021-06-18 by shantigilbert