EmuELEC/EmuELEC #579: Khadas VIM3 3705 FAN is not working, seems  i2c-tools is needed

eviltroooper, 2021-04-27:
i installed the 3705 FAN on my VIM3 but the Fan is not spinning..

there is a Fan.sh script.. but it seems i2c-tools need to add on busybox. is that possible?

https://github.com/hyphop/khadas-utils/blob/master/utils/fan

eviltroooper, 2021-04-27:
on Coreelec under Hardware there is "FAN" and i can enable it.. dont know how it works.. but it would be cool if someone can enable this on emuelec too.. :)

eviltroooper, 2021-04-29:
Mhh looks like it only use /sys/class/fan
Any idea how to change this on startup?

https://github.com/CoreELEC/service.coreelec.settings/blob/coreelec-19/src/resources/lib/modules/hardware.py#L646-L664

shantigilbert, 2021-04-29:
You can edit autostart.sh

eviltroooper, 2021-04-29:
Yeah I tested it with Autosparte.so..
Just need to remind that I need to edit it every update. 
don’t know if can can build a button on the Menu?

shantigilbert, 2021-04-29:
autostart.sh does not get removed on updates

eviltroooper, 2021-04-30:
For me it’s okay to use autostart.sh 
But don‘t know if autostart.sh will updated (if emulationstation will be updated), I will keep it in Mind.

If you want to build a switch it would be cool also..

closed 2021-04-30 by eviltroooper