EmuELEC/EmuELEC #580: error when build ecwolf@4.1

insmagic, 2021-04-27:
I have met the error when build ecwolf
it showed error as :
CMake Error at Locate SDL2.cmake
Please set SDL2_net_DIR to the location of sdl2_net-config.cmake.
then I get install with:
sudo apt install libsdl2-dev libsdl2-net-dev
It seems done.
when 4.0 I have not seen it.

shantigilbert, 2021-04-27:
So if you fixed it, what is the issue?

closed 2021-04-27 by shantigilbert

ChiaYen-Kan, 2021-05-04:
Hi 
i also build v4.1 fail on ecwolf 
the attach file is my build log, how to fix this error?
thank

[ecwolf_error.txt](https://github.com/EmuELEC/EmuELEC/files/6418880/ecwolf_error.txt)

insmagic, 2021-05-04:
> Hi 
> i also build v4.1 fail on ecwolf 
> the attach file is my build log, how to fix this error?
> thank
> 
> [ecwolf_error.txt](https://github.com/EmuELEC/EmuELEC/files/6418880/ecwolf_error.txt)
> 

sudo apt install libsdl2-dev libsdl2-net-dev
I used this command，your problem seems not the same as mine.Goodluck!

ChiaYen-Kan, 2021-05-05:
> sudo apt install libsdl2-dev libsdl2-net-dev
> I used this command，your problem seems not the same as mine.Goodluck!

i try the `sudo apt install libsdl2-dev libsdl2-net-dev`, but don't work for my case

miwasp, 2021-05-05:
Please try `sudo apt install libsdl2-mixer-dev libfluidsynth2`