EmuELEC/EmuELEC #581: Support for Selecting ROM Partition File System Type

lightvil, 2021-04-29:
**Is your feature request related to a problem? Please describe.**
I used 4.0 on Odroid N2 with a 256GB sdcard and Odroid Go Advance Rev 1.1. with a 64GB sdcard.
In 4.0 I modified emulstation-conf script and could mount the NTFS rom partition.
Just modified  "moun -t " option to ntfs.

Copying rom files is faster than VFAT filesystem.
I loved it.

But In 4.1 Its location moved from /starage/.config to SYSTEM.
I tried to modify it with squashfs tools but failed. It was not simple as 4.0.

I think for large SD Cards like 256GB VFAT is unstable and not efficient.
The NTFS file system is more efficient for storage utilization and performance than the VFAT file system.
The NTFS is supproted most system including Windows, Linux, MacOS(with extra software).

So I think it's very useful that a user can select file system types for a rom partition.

**Describe the solution you'd like**
1. Add an option like "rom_fs_type" to bootargs which defaults to 'vfat'
2. User can choose the file system type through this bootarg before booting the device.
2.  configure scripts can create and mount  rom partition using this bootarg from kernel command line parameters.

**Describe alternatives you've considered**
I think I can modify the configuration script(s) with some trials and errors.
But future updates will break this configruation resulting 

**Additional context**

shantigilbert, 2021-04-29:
This is not as easy as it sounds, the partition is made on first boot and its defined as vfat https://github.com/EmuELEC/EmuELEC/blob/bbb0bc0cca7be0e3aa36de574114098565b622d8/packages/sysutils/busybox/scripts/init#L695 , for this to work the setting needs to be defined BEFORE the first boot (and I am not even sure it will even work)

lightvil, 2021-04-30:
rom_fs_type='vfat'

> This is not as easy as it sounds, the partition is made on first boot and its defined as vfat [...]

OK, It seems that I found the right script. The "init"
I missed something about kernel parameters.

I'll try to find a way to give an information to "init" scripts.

lightvil, 2021-04-30:
@shantigilbert 

1. lets define a cumstom bootarg like this.
   `setenv bootargs "${initargs} ${displayopt} ${coreelec} rom_fs_type=ntfs"`
2. In the init sciript we can read it and define a file system type variable.  see line 1130 and below.  I just added a few lines...
```
ROM_FS_TYPE="vfat"
# parse command line arguments
for arg in $(cat /pr); do
  case $arg in
    rom_fs_type=ntfs)
      ROM_FS_TYPE="ntfs"
      ;;
    rom_fs_type=ext4)
      ROM_FS_TYPE="ext4"
      ;;
    BOOT_IMAGE=*)
      IMAGE_KERNEL="${arg#*=}"
      [ "${IMAGE_KERNEL:0:1}" = "/" ] && IMAGE_KERNEL="${IMAGE_KERNEL:1}"
      ;;
    SYSTEM_IMAGE=*)

```
3. And mount_roms() in init scripts, we can mount rom partition with ROM_FS_TYPE env. variable. (see line 690)
```
ount_roms() {
if /usr/bin/busybox mountpoint -q /storage ; then
    progress "Mounting roms"
  if [ -L "/storage/roms" ]; then
   /usr/bin/busybox rm /storage/roms > /dev/null 2>&1
  fi
    /usr/bin/busybox mkdir -p /storage/roms > /dev/null 2>&1
    mount -t ${ROM_FS_TYPE} "LABEL=EEROMS" /storage/roms > /dev/null 2>&1
```

4. fs-resize script can changed same way... see line 56
```
    ROM_FS_TYPE="vfat"
    PARTED_FS_TYPE="fat32"
    for arg in $(cat /proc/cmdline); do
      case $arg in
        rom_fs_type=ntfs)
          ROM_FS_TYPE="ntfs"
          PARTED_FS_TYPE="NTFS"
          ;;
        rom_fs_type=ext4)
          ROM_FS_TYPE="ext4"
          PARTED_FS_TYPE="ext2"
          ;;
      esac
    done
    
    StartProgress spinner "Creating EEROMS partition..." "parted -s -a optimal -m $DISK mkpart primary ${PARTED_FS_TYPE} 4296MB 100% &>/dev/null"
    partprobe &>/dev/null
    StartProgress spinner "Formatting EEROMS partition..." "mkfs.${ROM_FS_TYPE} -n EEROMS ${DISK}p3 &>/dev/null"
    StartProgress countdown "Rebooting in 5s...     " 5 "NOW"
```

If a user define rom_fs_type bootarg as ntfs, init script will create NTFS partition and mount it.

As newbie to EmuElec, I'm not familiar with using github and building EmuElec.
I'll try to build and test install with my Odroid N2.

I think this should work.

shantigilbert, 2021-04-30:
Thanks I will test it as soon as I get some free time and see if it works, seems like it should.

shantigilbert, 2021-05-03:
Fixed: in commit 11e6f71cd345107e94990046800cd3e129b187f6

lightvil, 2021-05-03:
@shantigilbert  The "ee_fstype" file in the BOOT partition, Good idea!

We can add more choices in the future in this way.
How about defining an emuelec boot configuration file named like 'emuelec-boot.ini' in BOOT partition.

BTW, Should I close this issue, right now?

shantigilbert, 2021-05-03:
> BTW, Should I close this issue, right now?

No, there is still one remaining issue, when using NTFS the update folder does not get mounted as the NTFS driver is not available on init, I need to set it up later on the boot process.

shantigilbert, 2021-05-12:
This change is only going to support exfat, fat32 and ext4, NTFS is not wrking as it should, there is no way to mount NTFS on init, and the .update folder will not be available for updates. Exfat should be the solution for you.

lightvil, 2021-06-01:
@shantigilbert I'm using Linux as Desktop OS. The ext4  will be good for me. Thank you.
I'll close this.

closed 2021-06-01 by lightvil