EmuELEC/EmuELEC #583: Odroid Go Super: 4.1 Stable - Game Exit HotKey not responding for non-retroarch systems

Desani, 2021-04-29:
**Describe the bug**
The Hotkey designated for quitting the game (1 + 6 quits game) does not respond outside of retroarch.

Currently I have tested the following systems: Scummvm, Doom, prDoom, PICO-8, PPSSPPSDL, SONIC 1 & 2 & CD, Drastic

**To Reproduce**
Steps to reproduce the behavior:
Run a game in any of the non-retroarch based emulators and attempt to quit using the hotkey 1+ 6. The system does not respond and you must quit by other means.

**EmuELEC version:**
 - 4.1 Stable - Hotkey Broken
 - 4.1-TEST-03302021 - Hotkey Broken
 - 4.1-TEST-02182021 - Hotkey works as expected

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - Brand: Odroid
 - Model Go Super

shantigilbert, 2021-06-07:
Can you try the latest test/beta?

zzeous, 2021-06-07:
I tested it on OGS.
Hotkey did not work when updated from 4.0 to 4.1.
but after formatting and installing as 4.1, hotkey was working normally.

shantigilbert, 2021-06-07:
Maybe you changed it by mistake probably? 

Can I close this issue now?

Desani, 2021-06-08:
Yeah, this can be closed now. The exit hotkey was changed for non-retroarch software to another key. After testing with the new HotKey, exit is working properly.

closed 2021-06-08 by Desani