EmuELEC/EmuELEC #589: Odroid Go Super: 4.1 Stable - Unable to play NDS/drastic

PikachuEXE, 2021-05-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Restore 4.0 stable backup SD card image
2. Update to 4.1 via SD card direct copy of tar file into `.update`
3. Boot again, update, reset binary, reconnect network, install drastic
4. Start a random game
5. See error

**Expected behavior**
- A game can be played
- Game saves/states retained

**Screenshots**
Error message:
[image]
[image]

**EmuELEC version:**
4.1

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: Rockchip
 - Brand: HardKernel
 - Model: Odroid-GO Advance

**Additional context**
I was trying to report game save being gone
But on 2nd try forward I just keep getting this error instead
Probably old config related coz I can play on clean install

Old config backup
[ee_backup_config.tar.gz](https://github.com/EmuELEC/EmuELEC/files/6409776/ee_backup_config.tar.gz)

I have no idea how to backup saves & states thus can't workaround via clean install + game restore

shantigilbert, 2021-05-02:
This is what the script keeps
      
      ```
        tar -zcvf "${BACKUPFILE}" /storage/.local/share/VVVVVV/* \
					  /storage/.cache/bluetooth/* \
					  /storage/.cache/timezone \
					  /storage/.config/scummvm/* \
					  /storage/.config/emuelec/configs/emuoptions.conf \
					  /storage/.config/emuelec/configs/emuelec.conf \
					  /storage/.emulationstation/es_*.cfg \
					  /storage/.emulationstation/themesettings \
					  /tmp/joypads/* \
					  /storage/.config/retroarch/*.cfg \
					  /storage/.config/ppsspp/* \
					  /storage/.config/retroarch/config/* \
					  /storage/.config/supertux* \
					  /storage/.emulationstation/scripts/drastic/config/* \
					  /storage/.emulationstation/scripts/drastic/*.dsv \
                      /storage/.config/emuelec/bin/pico-8/*
                      
```

As you can see drastic is there, unless you manually installed some where else?

PikachuEXE, 2021-05-03:
The game saves & save states are not backup

Drastic save folders:
- `/storage/.config/emulationstation/scripts/drastic/backup`
- `/storage/.config/emulationstation/scripts/drastic/savestates`

PikachuEXE, 2021-05-06:
Was trying to test on 4.1 but the script seems to be on some readonly partition?

PikachuEXE, 2021-05-07:
Keeping this opened until I am able to test the script myself

PikachuEXE, 2021-05-16:
Tested in `4.2-TEST-05142021`

PR #595 (2021-05-06): Fix drastic game saves & save states erased when resetting scripts & binaries to default
Fixes #589

Drastic save folders observed in EmuELEC `4.0`:
- `/storage/.config/emulationstation/scripts/drastic/backup`
- `/storage/.config/emulationstation/scripts/drastic/savestates`

[image]
[image]

Current drastic script part added in
https://github.com/EmuELEC/EmuELEC/blame/4f16bfbac1a5941671f0d050fcd2c29cf6822475/packages/sx05re/emuelec/config/emuelec/scripts/ee_backup.sh
is 14 months ago and might be for old drastic versions and thus folder structure

Note: My editor strips all trailing spaces aut [... cut]
closed 2021-05-16 by PikachuEXE