EmuELEC/EmuELEC #591: 4.1 stable on OGA no effect of shaders

sAKi2las, 2021-05-03:
No visible effect from shaders on different cores at 4.1 stable on OGA.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to core menu, view options, advanced system options
2. Click on shader set (is auto per default) set different shader
3. Select any shader ec "CRT/CRT-NESMINI:GLSLP
4. Return and restart system to take effect doesnt work

**Expected behavior**
No visible effect from shader

**EmuELEC version:**
 - Version 4.1 stable OGA

**Device (please complete the following information) and make sure its a [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices) **
 - SOC: G200513062627
 - Brand: hardkernel
 - Model ODROID-GO-Advance-BlackEdition
 - Version v2

shantigilbert, 2021-05-04:
Were they working before? I don't think shaders were working on the OGA

sAKi2las, 2021-05-05:
Shaders still work. 
1. Go to Games Settings
2. Set Smooth Games to "OFF"
3. Choose Shader

closed 2021-05-05 by sAKi2las