EmuELEC/EmuELEC #608: Playable to semi playable speeds on N64 on OGA?

Cee123, 2021-05-12:
This is not an issue, but sharing information. And I don't know where to post this. Just want to help out anyone else who has an Odroid Go Advance as I've had success at running quite a few harder to run games at playable speeds.

BUT I have managed to get Nintendo 64 at a playable to semi-playable (depending on the game) speeds on the Odroid Go Advance using Mupen64plus-next with GLES3! Most games run pretty decent. And it looks good! Less graphics bugs. Only downside is there's a bit of lag every now and then. This gives an alternative to using Parallel. This is a core I compiled using Ubuntu Linux. Some games will still run slow like Goldeneye.

[mupen64plus_next_libretro.zip](https://github.com/EmuELEC/EmuELEC/files/6465699/mupen64plus_next_libretro.zip)

The GLES3 core has a setting (that's not in GLES2 core) called Color buffer to RDRAM, and you can set that to TripleBuffer, restart, and the game will run smoother.

The settings I had to change under Quick Settings > Options:
4:3 Resolution: 320x240 (set this lowest)
16:9 Resolution: 640x360 (also set this lowest)
Hybrid Filter: Off (turn this one off!)
Framebuffer Emulation: On/Off (only turn this off on games where it doesn't cause issues)
Color Buffer to RDRAM: TripleBuffer (set this to TripleBuffer instead of Sync)
Cache Textures: Off (turn this off, eliminates a bit of lag)

Other things that help:
Turn off Vsync under Video Settings (makes a big difference for Banjo-Kazooie)
Overclock to 1.5GHz under EmuElec settings.

shantigilbert, 2021-05-12:
Thanks for sharing, this would be better suited in the forum http://emuelec.org

Cee123, 2021-05-13:
Not an issue just here if anyone's interested. I might share on the forum.

closed 2021-05-13 by Cee123