EmuELEC/EmuELEC #611: Wifi driver QCA6174 (X96 Max Plus Q2)

CharlyXCV, 2021-05-16:
Please, give support to the Wifi driver QCA6174. 

Is needed for the correct functionality of the X96 Max Plus Q2, which is the new generation of this famous S905X3 TV Box that changes its Wifi and Bluetooth connections. CoreELEC already done this in its last version. 

If you also work in the bluetooth connections –that also fails and I don't know if it's related– will be great. 

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

shantigilbert, 2021-05-17:
Someone already implemented them, but hes PR was not correct, waiting on the changes.

shantigilbert, 2021-05-20:
429c8bb6bfa65565488682de5b666bae5223b4f6

kakaxp, 2021-05-20:
> [429c8bb](https://github.com/EmuELEC/EmuELEC/commit/429c8bb6bfa65565488682de5b666bae5223b4f6)

How do I activate Wifi and BT? I do not understand

CharlyXCV, 2021-05-20:
I don’t understand it either. Can someone explain us how to fix it?

shantigilbert, 2021-05-20:
It means it has been added and will be available on next release (untested)

eric4gith, 2021-05-21:
I grabbed kevinrivm's code and built the image. I have an identical device.  WiFi & BT are working now. SSH through WiFi and DS3 through BT are both working. Just one minute simple testing.
Here is the log: http://ix.io/3nqA

I encountered error while building those additional drivers, so I made a few modification to make the building process smoother. 
I only modified /projects/Amlogic-ng/options in the line: 
ADDITIONAL_DRIVERS="$ADDITIONAL_DRIVERS gpu-aml-ng openvfd-driver wifi_dummy-aml media_modules-aml
                        ap6xxx-aml ssv6xxx-aml mt7668-aml RTL8189ES-aml RTL8189FS-aml RTL8723BS-aml RTL8814AU RTL8822BU-aml RTL8822BS-aml RTL8822CS-aml RTL8152-aml qca9377-aml qca6174-aml"

kakaxp, 2021-05-21:
> Peguei o código de kevinrivm e construí a imagem. Eu tenho um dispositivo idêntico. WiFi e BT estão funcionando agora. SSH por WiFi e DS3 por BT estão funcionando. Teste simples de apenas um minuto.
> Aqui está o log: http://ix.io/3nqA
> 
> Encontrei um erro ao construir esses drivers adicionais, então fiz algumas modificações para tornar o processo de construção mais suave.
> Eu apenas modifiquei / projects / Amlogic-ng / options na linha:
> ADDITIONAL_DRIVERS = "$ ADDITIONAL_DRIVERS gpu-aml-ng openvfd-driver wi-fi_dummy-aml
> media_modules-aml ap6xxx-aml ssv6xxx-aml mt7668-aml RTL8189ES-aml-aml-l8189ES-aml9 -aml RTL8814AU RTL8822BU-aml RTL8822BS-aml RTL8822CS-aml RTL8152-aml qca9377-aml qca6174-aml "

would it be possible to share in form of img so that we can test tb?  thanks

shantigilbert, 2021-05-21:
This driver is already implemented in the dev branch, no need to add it just pull latest changes if you are building it yourself

kakaxp, 2021-05-30:
Tks by update

closed 2021-06-18 by shantigilbert